Lora-net/LoRaMac-node #389: Sx1276 :not receive correct data in RX mode.

ios198, 2017-12-27:
I am trying to test Tx-Rx with SX1276.
This is my source code in TX for sending 'PING" continuously :
`   uint8_t Buffer[] = {'P', 'I', 'N', 'G'};
    uint8_t BufferPktLen = sizeof Buffer;
    uint8_t irqFlags = 0;

    Radio.Init( &RadioEvents );
    Radio.SetChannel( RF_FREQUENCY );
    Radio.SetTxConfig( MODEM_LORA, TX_OUTPUT_POWER, 0, LORA_BANDWIDTH,
                                 LORA_SPREADING_FACTOR, LORA_CODINGRATE,
                                 LORA_PREAMBLE_LENGTH, LORA_FIX_LENGTH_PAYLOAD_ON,
                                 true, 0, 0, LORA_IQ_INVERSION_ON, 3000 ); // Defines taken from the ping-pong source 
    Radio.Send( Buffer, BufferPktLen );
    while( 1 )
    {
        irqFlags = SX1276Read( REG_LR_IRQFLAGS );
        if( ( irqFlags & RFLR_IRQFLAGS_TXDONE_MASK ) == RFLR_IRQFLAGS_TXDONE )
        {
            // Tx Done
            Radio.Sleep( );
            DelayMs( 1 );
            Radio.Send( Buffer, BufferPktLen );
            // Toggle a pin to indicate that the transmission has finished.
        }
    }`
This is my code in RX to receive data:
`   Radio.Init( &RadioEvents );
    Radio.SetChannel( RF_FREQUENCY );
    uint8_t irqFlags = 0;
    Radio.SetRxConfig( MODEM_LORA, LORA_BANDWIDTH, LORA_SPREADING_FACTOR,
                                   LORA_CODINGRATE, 0, LORA_PREAMBLE_LENGTH,
                                   LORA_SYMBOL_TIMEOUT, LORA_FIX_LENGTH_PAYLOAD_ON,
                                   0, true, 0, 0, LORA_IQ_INVERSION_ON, true );
    Radio.Rx( RX_TIMEOUT_VALUE );
    while( 1 )
    {
        irqFlags = SX1276Read( REG_LR_IRQFLAGS );
        
        if( ( irqFlags & RFLR_IRQFLAGS_RXDONE_MASK ) == RFLR_IRQFLAGS_RXDONE )
        {
            // Rx Done
            Radio.Sleep( );
            // Toggle a pin to indicate that the transmission has finished.
             DelayMs( 1 );
             Radio.Rx( RX_TIMEOUT_VALUE );
        
        }
    }`
==>RxDone Interrupt occurred in RX when each 'PING' is sent from TX. However, received data is always 0.
Could you help me to solve this problem?

mluis1, 2018-02-05:
Have you solved your issue?
Can we close the issue?

closed 2018-02-08 by mluis1

ios198, 2018-02-08:
Yes i solved it.