Lora-net/LoRaMac-node #390: Sleep mode

gnob417, 2018-01-03:
Hi, can i stop RX mode or TX mode by using Radio.Sleep() ? 
and can i stop triggering interrupt by using TimerStop()?

mluis1, 2018-02-05:
Yes for both questions.

closed 2018-02-05 by mluis1

gnob417, 2018-02-06:
Did you do experiment about what i ask?
I use TimerStop() for rx1, rx2 before rx1 or rx2 starts, however, listen mode still triggered..

mluis1, 2018-02-06:
We are sorry but we don't really understand the issue that you are encountering or what you are trying to achieve.

As this issue is not really related to this project I would recommend that you try to ask your questions on the [LoRa Community](http://semtech.force.com/lora) portal.

But, trying to answer your question at least what we understand, if you set the radio to sleep mode in the middle of a transmission then it is normal that the frame is not received on the other side (the frame isn't received completely).
The radio should be put in sleep mode only once you have either a TxDone, RxDone, RxTimeout, RxError or TxTimeout.