Lora-net/LoRaMac-node #394: DutyCyled Regions, how to compute when the next transmission is possible ?

GrumpyOldPizza, 2018-01-17:
The current code (4.4.0) does compute dutyCycleTimeOff in ScheduleTx(), and then consequently delays the transmission.

If I now want to know ahead of time when the next transmission is possible, how would I go about that ?

Looks like the only place this is computed goes throu RegionNextChannel(), which also commits the channel to a TX operation, rather than just querying ...

mluis1, 2018-02-08:
We will try to add this functionality for the [Release Version 4.4.2](https://github.com/Lora-net/LoRaMac-node/milestone/2) milestone.

ZuratonE, 2018-07-02:
first call 
_CalculateBackOff( LastTxChannel );_
then call
_RegionCommonUpdateBandTimeOff( bool joined, bool dutyCycle, Band_t* bands, uint8_t nbBands )_,
which will return time to next tx

simonnilsson, 2019-08-26:
I made an attempt at adding this functionality in _LoRaMac.c_. I have taken the relevant parts from **ScheduleTx**. From my initial tests it seems to be working.
```
TimerTime_t LoraMacTimeToNextTx( )
{
    TimerTime_t dutyCycleTimeOff = 0;
    NextChanParams_t nextChan;

    // Update back-off
    CalculateBackOff( MacCtx.NvmCtx->LastTxChannel );

    nextChan.AggrTimeOff = MacCtx.NvmCtx->AggregatedTimeOff;
    nextChan.Datarate = MacCtx.NvmCtx->MacParams.ChannelsDatarate;
    nextChan.DutyCycleEnabled = MacCtx.NvmCtx->DutyCycleOn;
    if( MacCtx.NvmCtx->NetworkActivation == ACTIVATION_TYPE_NONE )
    {
        nextChan.Joined = false;
    }
    else
    {
        nextChan.Joined = true;
    }
    nextChan.LastAggrTx = MacCtx.NvmCtx->LastTxDoneTime;

    RegionNextChannel( MacCtx.NvmCtx->Region, &nextChan, &MacCtx.Channel, &dutyCycleTimeOff, &MacCtx.NvmCtx->AggregatedTimeOff );

    return dutyCycleTimeOff;
}
```

djaeckle, 2019-08-26:
Hi @simonnilsson,

many thanks for the proposal! We will double check it.

simonnilsson, 2019-08-26:
Hi @djaeckle,

After some consideration i think that **RegionNextChannel** probably shouldn't be used as it randomizes the next channel each time it's called. Calling **RegionCommonUpdateBandTimeOff** directly as suggested above is probably better, but has the drawback that you need to know the regions bands.

I have written an alternative to above with the requires the addition of a **RegionGetBands** method.
```c
TimerTime_t LoraMacTimeToNextTx( )
{
    TimerTime_t dutyCycleTimeOff = 0;
    Band_t *bands;
    uint8_t nbBands = 0;

    // Update back-off
    CalculateBackOff( MacCtx.NvmCtx->LastTxChannel );

    // Get region bands
    bands = RegionGetBands( MacCtx.NvmCtx->Region, &nbBands );

    dutyCycleTimeOff = RegionCommonUpdateBandTimeOff( 
        MacCtx.NvmCtx->NetworkActivation != ACTIVATION_TYPE_NONE,
        MacCtx.NvmCtx->DutyCycleOn,
        bands,
        nbBands
    );

    return dutyCycleTimeOff;
}
```

djaeckle, 2019-08-26:
HI @simonnilsson,

> After some consideration i think that RegionNextChannel probably shouldn't be used as it randomizes the next channel each time it's called.

I agree. Especially, this is not save for US based regions, as we the MAC has to be compliant to a channel sequence procedure - and, as you are already mentioned, it will change the floor plan.

I think you are on the correct track with your new proposal. But the problem i see is that the MAC calls the common region layer with information which is only available on each region directly.

I would recommend to add another function to the regional layer which is responsible to perform this procedure implemented on region individually. Furthermore, i think we need to take care of:
- Verification of enabled channels
- Verification of AggregatedTimeOff
- Datarate (with respect to ADR)

I guess we could reuse code parts of the function RegionXXNextChannel, just without any update of the floor plan.

Pirakutu, 2019-09-03:
I do something similar on the app side with help of the some information i get from the stack:

TimerTime_t LORA_TX_Delay ( uint8_t TxChannel, TimerTime_t TxTimeOnAir)
{
  TimerTime_t TRetVal = ( TimerTime_t )( -1 );
#if defined (REGION_EU868)
  MibRequestConfirm_t mibGet;
  LocalRegionEU868Params_t RegionInfo = {0};
  RegionCommonCalcBackOffParams_t BackoffParams = {0};

  memset(&mibGet, 0, sizeof(mibGet));
  mibGet.Type = MIB_NVM_CTXS;
  if( LoRaMacMibGetRequestConfirm( &mibGet ) == LORAMAC_STATUS_OK ){
    memcpy(&RegionInfo, mibGet.Param.Contexts->RegionNvmCtx, sizeof(RegionInfo));
  }

  BackoffParams.Channel = TxChannel;
  BackoffParams.Channels = RegionInfo.Channels;
  BackoffParams.DutyCycleEnabled = LoRaParamInit->DutyCycleOn;
  BackoffParams.Bands = RegionInfo.Bands;
  BackoffParams.Joined = LORA_JoinStatus();
  BackoffParams.TxTimeOnAir = TxTimeOnAir;

  RegionCommonCalcBackOff(&BackoffParams);

  for (uint8_t i = 0; i < MAX_NR_CHANNELS; i++)
  {
    if (RegionInfo.Channels[i].Frequency != 0){
      if( RegionInfo.Bands[RegionInfo.Channels[i].Band].TimeOff <= TimerGetElapsedTime( RegionInfo.Bands[RegionInfo.Channels[i].Band].LastTxDoneTime ) ){
        RegionInfo.Bands[RegionInfo.Channels[i].Band].TimeOff = 0;
        TRetVal = 0;
      }
      if(RegionInfo.Bands[RegionInfo.Channels[i].Band].TimeOff){
        TRetVal = MIN( RegionInfo.Bands[RegionInfo.Channels[i].Band].TimeOff - TimerGetElapsedTime( RegionInfo.Bands[RegionInfo.Channels[i].Band].LastTxDoneTime ),
                       TRetVal );
      }
    }
  }

#if defined (DEBUG)
  PRINTF("[debug] TX Delay Time -> %d ms\r\n", TRetVal);
#endif
#endif

  return TRetVal;
}

This code is run after every TX and feeds an APP timer, which is in charge of duty blocking on the app side.

MarkusPfundstein, 2019-10-21:
@djaeckle 
HI Djaeckle, thanks for your comment above. I did precisely this in my code and it yielded very good results for IN865, CN779, AU915, EU868 and US915.
Please see the code below.

```
932 LoRaMacStatus_t RegionEU868DutyCycleTimeOff(NextChanParams_t* nextChanParams, TimerTime_t *time)
933 {
934     uint8_t nbEnabledChannels = 0;
935     uint8_t enabledChannels[EU868_MAX_NB_CHANNELS] = { 0 };
936     uint8_t delayTx = 0;
937 
938     if( RegionCommonCountChannels( ChannelsMask, 0, 1 ) == 0 )
939     { // Reactivate default channels
940         ChannelsMask[0] |= LC( 1 ) + LC( 2 ) + LC( 3 ); 
941     }
942 
943     if ( nextChanParams->DutyCycleEnabled || !nextChanParams->Joined )
944     {
945         if (nextChanParams->AggrTimeOff <= TimerGetElapsedTime(nextChanParams->LastAggrTx)) 
946         {               
947             *time = RegionCommonUpdateBandTimeOff( nextChanParams->Joined, nextChanParams->DutyCycleEnabled, Bands, EU868_MAX_NB_BANDS );
948 
949             // count if we have any free channels. if yes, we don't need to wait for dutycycle
950             nbEnabledChannels = CountNbOfEnabledChannels( 
951                     nextChanParams->Joined,         
952                     nextChanParams->Datarate,       
953                     ChannelsMask,                   
954                     Channels,
955                     Bands,
956                     enabledChannels,
957                     &delayTx );
958             if (nbEnabledChannels > 0) {
959                 *time = 0;
960             } else {
961                 if (delayTx == 0) {
962                     *time = 0;
963                     return LORAMAC_STATUS_NO_CHANNEL_FOUND;
964                 }
965                 else
966                 {
967                     return LORAMAC_STATUS_DUTYCYCLE_RESTRICTED;
968                 }
969             }
970         }
971         else
972         {
973             *time = nextChanParams->AggrTimeOff - TimerGetElapsedTime( nextChanParams->LastAggrTx );
974             return LORAMAC_STATUS_DUTYCYCLE_RESTRICTED;
975         }
976     } else {
977         // no dutycycle enabled
978         *time = 0;
979     }
980     return LORAMAC_STATUS_OK;
981 }
```

The problem I have are regions KR920 & AS923. More specifically the (randomized) iteration through the enabledChannels and subsequent Radio calls to check if the channel is free are troublesome.

```
945     if( nbEnabledChannels > 0 )
946     {
947         for( uint8_t  i = 0, j = randr( 0, nbEnabledChannels - 1 ); i < AS923_MAX_NB_CHANNELS; i++ )
948         {
949             channelNext = enabledChannels[j];
950             j = ( j + 1 ) % nbEnabledChannels;
951 
952             // Perform carrier sense for AS923_CARRIER_SENSE_TIME
953             // If the channel is free, we can stop the LBT mechanism
954             if( Radio.IsChannelFree( MODEM_LORA, Channels[channelNext].Frequency, AS923_RSSI_FREE_TH, AS923_CARRIER_SENSE_TIME ) == true )
955             {
956                 // Free channel found
957                 *channel = channelNext;
958                 *time = 0;
959                 return LORAMAC_STATUS_OK;
960             }
961         }
962         return LORAMAC_STATUS_NO_FREE_CHANNEL_FOUND;
963     }

```

I wonder what the best way would be to use this randomized loop also in the DutyCycle functionality and to ensure that both functions (CalcDutyCycleBackoff & NextChannel) will hit the same channel. Or to make the calculation without hitting the Radio....

I also wonder why this loop is randomized. Can't it just loop through it from 0 to nbChannels -1 ? Probably to distribute the sensing a bit over the different channels right?

Greetings and thanks already for your input.

steph44fr, 2019-12-11:
@Pirakutu 
hello, I am interested in your solution but I can not link to 4.4.1 for the LocalRegionEU868Params_t type and the structure field mibGet.Param.Contexts
Do you have the definition of these types?
Thank you

Pirakutu, 2019-12-12:
@steph44fr  It is a lil more complicated, but it works also. To be able to get the active bands i had to add a function on the regionEU868.c/.h (actually added this to all regions):
```
void RegionEU868GetActiveBands ( Band_t ActiveBands[EU868_MAX_NB_BANDS])
{
  memcpy(ActiveBands, Bands, sizeof (Bands));
}
```
The delay function then looks something like this:

```
TimerTime_t LORA_TX_Delay ( uint8_t TxChannel, TimerTime_t TxTimeOnAir, ChannelParams_t Channels[MAX_NR_CHANNELS] )
{
  TimerTime_t TRetVal = ( TimerTime_t )( -1 );
#if defined (REGION_EU868)
  Band_t Bands[ EU868_MAX_NB_BANDS ] = {0};
  RegionCommonCalcBackOffParams_t BackoffParams = {0};

  RegionEU868GetActiveBands(Bands);

  BackoffParams.Channel = TxChannel;
  BackoffParams.Channels = Channels;
  BackoffParams.DutyCycleEnabled = LoRaParamInit->DutyCycleOn;
  BackoffParams.Bands = Bands;
  BackoffParams.Joined = LORA_JoinStatus();
  BackoffParams.TxTimeOnAir = TxTimeOnAir;

  RegionCommonCalcBackOff(&BackoffParams);

  for (uint8_t i = 0; i < MAX_NR_CHANNELS; i++)
  {
    if (Channels[i].Frequency != 0){
      if( Bands[Channels[i].Band].TimeOff <= TimerGetElapsedTime( Bands[Channels[i].Band].LastTxDoneTime ) ){
        Bands[Channels[i].Band].TimeOff = 0;
        TRetVal = 0;
      }
      if(Bands[Channels[i].Band].TimeOff){
        TRetVal = MIN( Bands[Channels[i].Band].TimeOff - TimerGetElapsedTime( Bands[Channels[i].Band].LastTxDoneTime ),
                       TRetVal );
      }
    }
  }

#if defined (DEBUG)
  PRINTF("[debug] TX Delay Time -> %d ms\r\n", TRetVal);
#endif
#endif

  return TRetVal;
}
```

The parameter channels can be extracted with mib commands (MIB_CHANNELS, MIB_CHANNELS_MASK). I usually do this after having sent a message (inside McpsConfirm function). TxChannel and TxTimeOnAir are also stuff you get from an McpsConfirm. Hope this helps.

Cheers,

closed 2020-02-18 by mluis1