Lora-net/LoRaMac-node #396: Some problem about OnMacStateCheckTimerEvent after PrepareRxDoneAbort

suyiii, 2018-01-19:
Hello!

Recently, I tested the project and found that it's very easy to get into a dead loop while excuting OnMacStateCheckTimerEvent after PrepareRxDoneAbort. 

I sent Confirmed packge and got into PrepareRxDoneAbort. Then I came to OnMacStateCheckTimerEvent. In this function, the AckTimeoutRetriesCounter was less than  MAX_ACK_RETRIES, so I excuted ScheduleTx() . LoRaMacFlags.Bits.McpsInd was 1 and I had to open rx2 at the end of OnMacStateCheckTimerEvent. And than I didn't get TxDone signal.

Do I need to set LoRaMacFlags.Bits.McpsInd to 0? I found that the problem will be solved if I do like this.And do I lose anything?

Thank you for your help

djaeckle, 2018-02-08:
Hi suyiii,

thanks for your report. From your description, it seems like you have changed the code of the MAC layer. Could you please provide the details of your adaptions?

suyiii, 2018-02-08:
Thank you for your response.

I guess I've found the problem.

Indeed，I added Radio.Standby( ); before I opened rx windows, which will affect RxWindowSetup(). It means that every time I use OnRxWindow1TimerEvent or OnRxWindow1TimerEvent to open windows, the window will restart no matter what the radio is in. 

If I delete Radio.Standby( ), the radio will not open rx2 window because the Radio.GetStatus( ) == RF_IDLE is not true. Right?

djaeckle, 2018-02-08:
Which version are you using?

mluis1, 2018-02-23:
As there is no activity we decide to close this issue.

closed 2018-02-23 by mluis1