Lora-net/LoRaMac-node #399: Incorrect next beacon channel after missed beacon during a reacquisition

snelson-senet, 2018-01-30:
On a missed beacon event in  **beacon reaquisition state**,  calculated next beacon time is the same as the current missed beacon time, causing a second immediate beacon listen and guaranteed miss on beacon channel+1. This 2nd listen & miss results in BeaconCtx.BeaconTime being incremented by the beacon interval again and all subsequent beacon channel calculations  are off by +1.

Here is a trace of the failure.  First column is TimerGetCurrentTime() timestamp.

**1 Beacon lock then eventual beacon miss leading to beacon reacquistion state**
86489509    Beacon Locked:  BeaconTime=1201373696                                         
86617526    Beacon Missed

**2 First reacquisition beacon miss**
86744874    Beacon Listen Frequency=926900000 
86745005    Radio Rx Timeout 
86745006    OnClassBTimerEvent() - BEACON_REACQUISITION **beaconEventTime=60!**

**3 Followed quickly by the next beacon listen and miss** 
86745071    Beacon Listen Frequency=927500000
86745333    Radio Rx Timeout 
86748344    Beacon Missed 

The mysterious second beacon listen is due to the 60 ms next beacon time miscalculation in 2. How did this happen? Well, let's take a look at beacon time calculation in BEACON_STATE_REACQUISTION

`beaconEventTime = ( ( currentTime - BeaconCtx.LastBeaconRx ) % BeaconCtx.Cfg.Interval );
  beaconEventTime = BeaconCtx.Cfg.Interval - beaconEventTime
  // Take window enlargement into account
  beaconEventTime -= ( ( BeaconCtx.ListenTime * BeaconCtx.Cfg.SymbolToExpansionFactor ) >> 1);`

Enlarging the window causes the first reacquistion radio rx timeout (2) to occur before the actual expected beacon time, so the next time calculation is for the same beacon, but now configured to listen on beacon channel + 1.

As a quick and dirty test, I removed the last line in the code snippet and so far so good in that after a reacq miss the  spurious listens no longer occur and the next beacon channel calculations are correct.

djaeckle, 2018-02-06:
Hi snelson-senet,

thank you very much for your report and the detailed information. I will try to reproduce the issue on my side. I need some time for the verification.

djaeckle, 2018-02-06:
Hi snelson-senet,

could you tell me which commit was the base for your tests? Thanks in advance.

snelson-senet, 2018-02-06:
Hi djaeckle,

I am working with a merge of feature/class_b commit 258fcc and the v4.4.0 tag commit

djaeckle, 2018-02-19:
Hi snelson-senet,

we have reviewed the implementation and we think it has an issue. In the current implementation, the window movement will be applied to the BeaconCtx.NextBeaconRx variable. Actually, it should not be applied to it as an increment, but as an assignment instead. Otherwise, the error will be propagated with every beacon miss. Also, we will update the computation of the BeaconCtx.ListenTime.

We are testing a fix for it currently. We will apply it to the feature/class_b branch as soon as possible.

x893, 2018-02-19:
Hi,
can you recommend network server (standalone or public) for test CLASS B (with currect /class_b branch) ?
Thanks in advance

djaeckle, 2018-02-23:
Hi snelson-senet,

we have provided a set of updates for the feature/class_b branch. We have fixed the issue with the error propagation when a sequence of beacons has been missed. Furthermore, we have done some refactoring and simplifications. Please let me know if your issue has been solved.

djaeckle, 2018-02-23:
Hi x893,

please open up a new issue for your question as it is our of topic for this issue.

snelson-senet, 2018-02-23:
Hi djaeckle,

Thanks for the update. I'll test this early next week and report back my results.

mluis1, 2018-03-16:
We suppose that everything is working properly now.
If you don't mind we close the issue.

closed 2018-03-16 by mluis1