Lora-net/LoRaMac-node #400: Incorrect beacon channel after beacon CRC0 error

snelson-senet, 2018-01-31:
After receiving a beacon that fails crc0 but passes crc1 validation, subsequent beacon channels are off by -1.  In LoRaMacClassBRxBeacon(),  beaconReceived flag is set true if either crc0/crc1 check passes.  The problem is that if  crc0 check fails BeaconCtx.BeaconTime is not updated, causing subsequent beacon channel calculations to be incorrect.  

I changed beaconReceived to be set  true only if both crc validations pass. Now the failure condition results in a beacon miss event which updates BeaconCtx.BeaconTime by the beacon interval to keep next beacon channel calculations in sync with actual channel.

djaeckle, 2018-02-01:
Hi snelson-senet,

thanks for your report. Could you provide the details of your changes?

snelson-senet, 2018-02-01:
Hi djaeckle,

I just created a pull-request with my change.  This may not be the optimal solution, but it's my workaround for now until the patch.

djaeckle, 2018-02-01:
Hi snelson-senet,

many thanks for the pull request! I will take a look at it.

djaeckle, 2018-02-06:
Hi snelson-senet,

thanks for the report. We discussed your pull request and agree with the solution. Actually, this is not only related to the issue which you described, but also a different one:

The beacon frame is valid if crc0 is correct - it is even valid if we have an invalid crc1. The reason for this is that LoRaWAN gateways will send beacons simultaneously. The first part of the frame which is protected with crc0 is similar on all transmissions. The second part of the frame is gateway specific and could be different on every gateway.

Since the frame transmission overlaps, it could happen that we get an invalid crc1, if the radio was not able to demodulate the second part of the frame.

We will apply your pull request asap.

snelson-senet, 2018-02-06:
Great! I am glad I could help.

closed 2018-02-09 by mluis1