Lora-net/LoRaMac-node #404: Select sub-band for Gateway 8 channels

sabas1080, 2018-02-01:
Hi

I'm trying to connect to a multitech gateway that is listening in sub-band 7, I've seen some examples but I still can not understand how it works:

`static uint16_t multitechGatewayChannelsMask[6] = { 0xFF, 0x00, 0x00, 0x00, 0x00, 0x00};`

Example:

`static uint16_t multitechGatewayChannelsMask[6] = { 0xFF, 0x00, 0x00, 0x00, 0x00, 0x00};`

 ```   mibReq.Type = MIB_CHANNELS_DEFAULT_MASK;
    mibReq.Param.ChannelsDefaultMask = multitechGatewayChannelsMask;
    LoRaMacMibSetRequestConfirm( &mibReq );
   
    mibReq.Type = MIB_CHANNELS_MASK;
    mibReq.Param.ChannelsMask = multitechGatewayChannelsMask;
    LoRaMacMibSetRequestConfirm( &mibReq );`

Thanks

jamesl-dm, 2018-02-02:
Don't you need:
static uint16_t multitechGatewayChannelsMask[6] = { 0x0000, 0x0000, 0x00FF, 0x0000, 0x0000, 0x0000};
?

sabas1080, 2018-02-02:
Ok,

`static uint16_t multitechGatewayChannelsMask[6] = { 0x0000, 0x0000, 0x0000, 0x00FF, 0x0000, 0x0100};`

 for Subband 7 with channel of 500khz?

closed 2018-02-05 by mluis1