Lora-net/LoRaMac-node #405: JOIN TX CHAN SEQ V4.4.0

rrattinypcbs, 2018-02-05:
Checking for random channel assignment, the high band is probed at much higher rate than lower bands. A few samples follow. Testing lab is questioning this and saying that it should be randomized across all 72 channels. The 1.0.2 regional parameters for US915 do not state equally distributed. Any one had a tester question this before? What was the answer? 
v/r,
Rob
TX CH:71, PWR:0, BW:2, on freq 914200000 Hz at DR 4
TX CH:12, PWR:0, BW:0, on freq 904700000 Hz at DR 0
TX CH:70, PWR:0, BW:2, on freq 912600000 Hz at DR 4
TX CH:66, PWR:0, BW:2, on freq 906200000 Hz at DR 4
TX CH:0, PWR:0, BW:0, on freq 902300000 Hz at DR 0
TX CH:65, PWR:0, BW:2, on freq 904600000 Hz at DR 4
TX CH:65, PWR:0, BW:2, on freq 904600000 Hz at DR 4
TX CH:34, PWR:0, BW:0, on freq 909100000 Hz at DR 0
TX CH:71, PWR:0, BW:2, on freq 914200000 Hz at DR 4
TX CH:68, PWR:0, BW:2, on freq 909400000 Hz at DR 4
TX CH:6, PWR:0, BW:0, on freq 903500000 Hz at DR 0
TX CH:66, PWR:0, BW:2, on freq 906200000 Hz at DR 4
TX CH:67, PWR:0, BW:2, on freq 907800000 Hz at DR 4
TX CH:61, PWR:0, BW:0, on freq 914500000 Hz at DR 0
TX CH:66, PWR:0, BW:2, on freq 906200000 Hz at DR 4
TX CH:71, PWR:0, BW:2, on freq 914200000 Hz at DR 4
TX CH:16, PWR:0, BW:0, on freq 905500000 Hz at DR 0
TX CH:69, PWR:0, BW:2, on freq 911000000 Hz at DR 4
TX CH:70, PWR:0, BW:2, on freq 912600000 Hz at DR 4
TX CH:9, PWR:0, BW:0, on freq 904100000 Hz at DR 0
TX CH:68, PWR:0, BW:2, on freq 909400000 Hz at DR 4
TX CH:65, PWR:0, BW:2, on freq 904600000 Hz at DR 4
TX CH:25, PWR:0, BW:0, on freq 907300000 Hz at DR 0
TX CH:70, PWR:0, BW:2, on freq 912600000 Hz at DR 4
TX CH:71, PWR:0, BW:2, on freq 914200000 Hz at DR 4
TX CH:5, PWR:0, BW:0, on freq 903300000 Hz at DR 0
TX CH:69, PWR:0, BW:2, on freq 911000000 Hz at DR 4
TX CH:67, PWR:0, BW:2, on freq 907800000 Hz at DR 4
TX CH:10, PWR:0, BW:0, on freq 904300000 Hz at DR 0
TX CH:64, PWR:0, BW:2, on freq 903000000 Hz at DR 4
TX CH:65, PWR:0, BW:2, on freq 904600000 Hz at DR 4
TX CH:24, PWR:0, BW:0, on freq 907100000 Hz at DR 0
TX CH:71, PWR:0, BW:2, on freq 914200000 Hz at DR 4
TX CH:68, PWR:0, BW:2, on freq 909400000 Hz at DR 4
TX CH:60, PWR:0, BW:0, on freq 914300000 Hz at DR 0
TX CH:66, PWR:0, BW:2, on freq 906200000 Hz at DR 4
TX CH:68, PWR:0, BW:2, on freq 909400000 Hz at DR 4
TX CH:20, PWR:0, BW:0, on freq 906300000 Hz at DR 0
TX CH:66, PWR:0, BW:2, on freq 906200000 Hz at DR 4
TX CH:70, PWR:0, BW:2, on freq 912600000 Hz at DR 4
TX CH:55, PWR:0, BW:0, on freq 913300000 Hz at DR 0
TX CH:69, PWR:0, BW:2, on freq 911000000 Hz at DR 4
TX CH:69, PWR:0, BW:2, on freq 911000000 Hz at DR 4
TX CH:44, PWR:0, BW:0, on freq 911100000 Hz at DR 0
TX CH:67, PWR:0, BW:2, on freq 907800000 Hz at DR 4
TX CH:64, PWR:0, BW:2, on freq 903000000 Hz at DR 4
TX CH:63, PWR:0, BW:0, on freq 914900000 Hz at DR 0
TX CH:71, PWR:0, BW:2, on freq 914200000 Hz at DR 4
TX CH:71, PWR:0, BW:2, on freq 914200000 Hz at DR 4
TX CH:57, PWR:0, BW:0, on freq 913700000 Hz at DR 0
TX CH:69, PWR:0, BW:2, on freq 911000000 Hz at DR 4
TX CH:66, PWR:0, BW:2, on freq 906200000 Hz at DR 4
TX CH:47, PWR:0, BW:0, on freq 911700000 Hz at DR 0
TX CH:65, PWR:0, BW:2, on freq 904600000 Hz at DR 4
TX CH:66, PWR:0, BW:2, on freq 906200000 Hz at DR 4
TX CH:19, PWR:0, BW:0, on freq 906100000 Hz at DR 0
TX CH:71, PWR:0, BW:2, on freq 914200000 Hz at DR 4
TX CH:68, PWR:0, BW:2, on freq 909400000 Hz at DR 4
TX CH:36, PWR:0, BW:0, on freq 909500000 Hz at DR 0
TX CH:67, PWR:0, BW:2, on freq 907800000 Hz at DR 4

mluis1, 2018-02-06:
The DR4 datarate can only be randomized over 8 channels. The 500 kHz channels. The restrictions for these channels are different from the 125 kHz channels which must be randomized over the 72 channels and each channel must be used before restarting the operation. Please refer to the "LoRaWAN Regional Parameters" document.

Something we don't understand in your log is why sometimes you have DR0. The log you presented doesn't look like to be generated by only one device. We should either see all lines with DR0 or DR4 not a mix of the 2 datarates.

I don't know which testing lab provided this answer but, the current Stack successfully gets the LoRa-Alliance certification and has been tested by all official test houses.

Some of the certification tests verify the channel plan usage and they all succeed.

mluis1, 2018-02-06:
Sorry, we guess that we have misunderstood the issue. We thought the log was for normal operation but in reality it is for the Join procedure.

Indeed you may have an issue with your implementation. The datarate should alternate between DR0 and DR4 every time.

The current implementation correctly handles this and it has been verified by the different test houses.
https://github.com/Lora-net/LoRaMac-node/blob/f42be67be402a40b3586724800771bfe13fb18e6/src/mac/region/RegionUS915.c#L736

Please note that the other comments on the previous answer are still valid.

rrattinypcbs, 2018-02-06:
Thanks, this is for the V1.1.2 cube expansion software (Keil uV5) with the B-L072Z-LRWAN1 ST discovery board. I am working to create the board package now to make it easier for test and use same tools as you.

closed 2018-02-06 by rrattinypcbs