Lora-net/LoRaMac-node #407: LinkAdrReq command in US915 and US915 Hybrid

Adamzat, 2018-02-13:
Hi every one, 
I am using the 4.4.0 stack, and i did many tests for LinkAdrReq command which work very well. However, i have a strange behavior (in LinkAdrAns) regarding some specific commands in US915 Hybrid. Below some examples:

1) "LinkAdrReq" = 0x0345000061": my device retruns an LinAdrAns with : 0x0307
2) "LinkAdrReq" = 0x0305010071" my device retruns an LinAdrAns with: 0x0304
3) "LinkAdrReq"= 0x0315010071" my device retruns an LinAdrAns with: 0x0304
4) "LinkAdrReq"= 0x0325010071" my device retruns an LinAdrAns with: 0x0304
5) "LinkAdrReq" = 0x0335010071" my device retruns an LinAdrAns with: 0x0304

Please, do you think that is normal? Because i think that the correct LinkAdrAns should be:
"0x0305" for all 5 examples?

Thank you for your help.

AdamZat

djaeckle, 2018-03-05:
Hi Adamzat,

thanks for the report. We will analyze it. I will come back to you soon.

tortizactility, 2018-07-18:
@Adamzat 

After decoding your MAC commands:

```MACCMD: (FOPTS) 0345000070
MACCMD: 03 ( LinkADRReq ), TXPower: 5 (6dBm), DataRate: DR4 (SF8/BW125KHz), ChMask: 0x0000, NbRep: 0, ChMaskCntl: 7 (RFU)
MACCMD: (FOPTS) 0307
MACCMD: 03 ( LinkADRAns ), Status: 0x07, Channel mask: ACK, Data rate: ACK, Power: ACK

MACCMD: (FOPTS) 0305010071
MACCMD: 03 ( LinkADRReq ), TXPower: 5 (6dBm), DataRate: DR0 (SF12/BW125KHz), ChMask: 0x0001, NbRep: 1, ChMaskCntl: 7 (RFU)
MACCMD: (FOPTS) 0304
MACCMD: 03 ( LinkADRAns ), Status: 0x04, Channel mask: NACK, Data rate: NACK, Power: ACK

MACCMD: (FOPTS) 0315010071
MACCMD: 03 ( LinkADRReq ), TXPower: 5 (6dBm), DataRate: DR1 (SF11/BW125KHz), ChMask: 0x0001, NbRep: 1, ChMaskCntl: 7 (RFU)
MACCMD: (FOPTS) 0304
MACCMD: 03 ( LinkADRAns ), Status: 0x04, Channel mask: NACK, Data rate: NACK, Power: ACK

MACCMD: (FOPTS) 0325010071
MACCMD: 03 ( LinkADRReq ), TXPower: 5 (6dBm), DataRate: DR2 (SF10/BW125KHz), ChMask: 0x0001, NbRep: 1, ChMaskCntl: 7 (RFU)
MACCMD: (FOPTS) 0304
MACCMD: 03 ( LinkADRAns ), Status: 0x04, Channel mask: NACK, Data rate: NACK, Power: ACK

MACCMD: (FOPTS) 0335010071
MACCMD: 03 ( LinkADRReq ), TXPower: 5 (6dBm), DataRate: DR3 (SF9/BW125KHz), ChMask: 0x0001, NbRep: 1, ChMaskCntl: 7 (RFU)
MACCMD: (FOPTS) 0304
MACCMD: 03 ( LinkADRAns ), Status: 0x04, Channel mask: NACK, Data rate: NACK, Power: ACK
```

For me, the beahvior seems correct. The usage of ChMaskCntl:7 means "**All 125 kHz OFF, ChMask applies to channels 64 to 71**"
[image]

In addition of  ChMaskCntl:7, you are trying to use the DataRate linked to 125kHz (disabled by the ChMaskCntl value) : DR0-DR1-DR2. It's not possible so the stack replied NACK for DataRate and ChMask

[image]

Only the DR4 is accepted by the device because it's a 500kHz configuration and it's compatible with ChMaskCntl value.

closed 2018-07-26 by mluis1