Lora-net/LoRaMac-node #409: Class B: PingSlotInfoReq (1.0.2) or PingSlotInfoReq (1.1)?

brocaar, 2018-02-21:
I've been working on implementing Class-B (server-side) using the https://github.com/Lora-net/LoRaMac-node/tree/feature/class_b for testing and all works fine.

However I have one comment:

The code seems to be implementing the `PingSlotInfoReq` mac-command from LoRaWAN 1.1, which is binary incompatible with the `PingSlotInfoReq` from LoRaWAN 1.0.2. For now I've just implemented the LoRaWAN 1.1 version and I would be happy to just forget about the LoRaWAN 1.0.2 version of this mac-command. However, I just want to be sure if this is the right approach and if it was intended to only implement the LoRaWAN 1.1 version at your side.

### LoRaWAN 1.1

[image]

### LoRaWAN 1.0.2

[image]

mluis1, 2018-02-21:
We are implementing the feature/class_b branch against the LoRaWAN 1.0.3 specification.
The differences between 1.0.3 and 1.0.2 are:

* This specification is the one where the LoRa Alliance has officialized the ClassB specification.  
The ClassB specification chapter on LoRaWAN 1.0.3 is exactly the same as the one on LoRaWAN1.1
* The LoRaWAN 1.0.3 also adds the DeviceTimeReq MAC command as replacement for BeaconTimingReq (deprecated).

brocaar, 2018-02-21:
That is great, less things to worry about from my side 👍 

As a note: I will add support for Class-B to https://www.loraserver.io really soon :-) (I've got it already working locally including the `DeviceTimeReq` mac-command using the latest version of your class_b code)

closed 2018-02-21 by brocaar

1405224, 2018-05-15:
Hi, may i know what platform did you used? and how can i buy one?
Thanks

1405224, 2018-05-18:
Hi,
I have followed the instructions to implement this project and now I'm facing the errors below:
First when i build, I get:
[build] make[2]: *** [src/apps/LoRaMac/LoRaMac-classA] Error 1
[build] make[1]: *** [src/apps/LoRaMac/CMakeFiles/LoRaMac-classA.dir/all] Error 2
[build] make: *** [all] Error 2
[build] Build finished with exit code 2

Then, when i try to Debug, i get a popup message:
launch: program '/Users/LoRaMac-node/build/src/apps/LoRaMac/LoRaMac-classA' does not exist

Also, in the 'Problems' tap I've got:
Property MIDebuggerPath is not allowed

Could anyone please help me??

Thanks