Lora-net/LoRaMac-node #411: Incorrect LoRa Bandwidth Value Limitation?

gojimmypi, 2018-02-27:
There's a comment in sx1276.c claiming 

> // REMARK: When using LoRa modem only bandwidths 125, 250 and 500 kHz are supported

at this line: https://github.com/Lora-net/LoRaMac-node/blob/master/src/radio/sx1276/sx1276.c#L711

however the datasheet shows valid values for LoRa RegModemConfig1 (page 112) for all these bandwidth values:

[image]

and even this specification table on page 19 listed power for each LoRa receiver bandwidth value::

[image]

I was wondering where the limitation of 125, 250, and 500 kHz is documented?

Thanks

GuntherSchulz, 2018-02-28:
This body of work in an implementation of LoRaWan, as specified by the LoRa Alliance, which dictates what physical communication parameters are used to be able to work with the Public LoRaWan Network. 

As such, it implements what is needed to be compliant with their specification, and nothing more. More directly, any setting that are NOT in line with with the spec are non-compliant, and have no place in LoRaWan. Moreover, any non-compliant (read not-otherwise specified) physical setups will simply be ignored by the public WAN.

What you are referring to, is the physical capabilities of the Radio Transceiver in question, (which of course you are free to roll out for your own purposes in any way you fit, provided it meets local regulations, of course). Just don't hope your custom setup will in any way be even seen by the LoRaWan.

Hope this clears it up for you :)

nestorayuso, 2018-02-28:
Yes, LoRa has bandwidths lower than 125KHz, but LoRaWAN doesn't use them.
And this repository if for LoRaWAN

see https://github.com/Lora-net/LoRaMac-node/pull/264

closed 2018-02-28 by mluis1

gojimmypi, 2018-02-28:
@nestorayuso and particularly @GuntherSchulz - thanks for taking the time for your detailed response. :) I was not aware that the LoRaWAN spec had this requirement. I've been working on the RadioHead drivers (LoRa, no WAN) when I found this repository.

Reference for future viewers:

http://www.airspayce.com/mikem/arduino/RadioHead/

https://www.rs-online.com/designspark/rel-assets/ds-assets/uploads/knowledge-items/application-notes-for-the-internet-of-things/LoRaWAN%20Specification%201R0.pdf

From page 40:

[image]