Lora-net/LoRaMac-node #413: LoRa Stack 4.4.0 and AS923 Dwell Problem

ARAradtec, 2018-03-01:
The default Dwell uplink and downlink in AS923 is on. and since there is a max payload check now in the stack Rx Done function, commands received in the RX windows from servers that were configured with dwell off, are longer then the allowed dwell on max payload and are discarded, including the TX Parameters Setup request command (that sets the dwell off ). (i am using multitech AEP MTCDT-H5-210A Firmware 1.4.3).
setting the define **AS923_DEFAULT_DOWNLINK_DWELL_TIME** to 0 (off) solves the problem, and maybe it is a better default value for the end unit, because it allows the server to set the downlink dwell value for new joined units according to its configuration, and prevent the need to preconfigure the end unit dwell Values ,to match the server or the network dwell configuration.

dsmith-tek, 2018-03-07:
If you can guarantee that your sensor will never operate in a region with the Dwell Time restriction then I don't see any issue with you changing the default value.

However, I don't think this makes sense to incorporate into the Region as a whole. Since the AS923 region runs in multiple different countries it should, by default, ensure that it is compliant with all local regulations.  While some of these countries don't require the dwell time limitation, others do so it needs to respect that unless instructed otherwise.

I would argue however that it seems strange that the sensor would apply the dwell time restriction to packets it has already received, and validated. It is not the job of an end-device to police other devices dwell times, and I think it is reasonable for the device to assume that the network knows what it is doing.

ARAradtec, 2018-03-08:
Please notice that i have changed the down link dwell only.  the up link dwell stayed on. so my  unit will keep the dwell restriction at start, but will except all packets. this way the server is able to set both up and down dwells with the TX Parameters Setup command.

mluis1, 2018-07-02:
No activity we close this issue.

closed 2018-07-02 by mluis1