Lora-net/LoRaMac-node #418: FSK / GFSK SX1272/SX1276

GrumpyOldPizza, 2018-03-12:
As per https://github.com/Lora-net/LoRaMac-node/issues/417, EU868 DR_7 should use GFSK with BT 0.5. The SX1272/SX1276 code uses the default register settings, which leads to FSK usage instead of GFSK.

Furthermore, there is this here in https://github.com/Lora-net/lora_gateway/blob/master/libloragw/src/loragw_radio.c:

    /* Set in FSK mode */
    x = lgw_sx127x_reg_w(SX1276_REG_OPMODE, 0);
    wait_ms(100);
    x |= lgw_sx127x_reg_w(SX1276_REG_OPMODE, 0 | (ModulationShaping << 3)); /* Sleep mode, no FSK shaping */
    wait_ms(100);
    x |= lgw_sx127x_reg_w(SX1276_REG_OPMODE, 1 | (ModulationShaping << 3)); /* Standby mode, no FSK shaping */
    wait_ms(100);

So will a if you change SX1272/SX1276 from FSK to GFSK BT 0.5, will that break then gateways as well ?

mluis1, 2018-03-12:
This setting doesn't break anything. As said in issue #417 there is no real impact.

GrumpyOldPizza, 2018-03-12:
Would it make sense to properly update the standard to explicitly say GFSK BT 0.5 ? (I am a software engineer with not a lot of deep RF knownledge).

mluis1, 2018-03-12:
Yes, I also think that it would be a nice idea to update the standard with this kind off settings.

mluis1, 2018-03-28:
In a future version of the LoRaWAN Regional Parameters document the modulation shaping is to be set to 1.
Thus we reverted back the commit b137f8633c6fccb7027dc703c3193dd5ee8645c1

closed 2018-03-28 by mluis1

GrumpyOldPizza, 2018-03-28:
Does this apply to SX1276 as well ?

mluis1, 2018-03-28:
Yes, it applies to all radios.