Lora-net/LoRaMac-node #419: Bad appNonce size ?

exanos, 2018-03-12:
Could you explain me why memcpy copies 6 bytes whereas appNonce size is 3 ?

Thanks

https://github.com/Lora-net/LoRaMac-node/blob/1cdd9ccec4c9f05b616e7112059be4a9e358c571/src/mac/LoRaMacCrypto.c#L202

mluis1, 2018-03-12:
To derive the session keys we need the AppNonce (3 bytes) and the NetID (3bytes)

> NwkSKey = aes128_encrypt(AppKey, 0x01 | AppNonce | NetID | DevNonce | pad16)
AppSKey = aes128_encrypt(AppKey, 0x02 | AppNonce | NetID | DevNonce | pad16)

The *appNonce variable points to AppNonce index of the JoinAccept message buffer. Thus, by copying 6 bytes we are copying in one step the AppNonce and the NetID.

exanos, 2018-03-12:
Thanks @mluis1 !

closed 2018-03-12 by exanos