Lora-net/LoRaMac-node #425: ClassC retransmission issue ?

DoubleCheeseBacon, 2018-03-21:
Hi all,

My device is using C class, LoRaMacParams.ChannelsNbRep = 3, sending unconfirmed Tx.

static void OnMacStateCheckTimerEvent( void )
{
        .......

        if( ( NodeAckRequested == false ) && ( txTimeout == false ) )
        {
            if( ( LoRaMacFlags.Bits.MlmeReq == 1 ) || ( ( LoRaMacFlags.Bits.McpsReq == 1 ) ) )
            {
                if( ( LoRaMacFlags.Bits.MlmeReq == 1 ) && ( MlmeConfirm.MlmeRequest == MLME_JOIN ) )
                {// Procedure for the join request
                    MlmeConfirm.NbRetries = JoinRequestTrials;

                    if( MlmeConfirm.Status == LORAMAC_EVENT_INFO_STATUS_OK )
                    {// Node joined successfully
                        UpLinkCounter = 0;
                        ChannelsNbRepCounter = 0;
                        LoRaMacState &= ~LORAMAC_TX_RUNNING;
                    }
                    else
                    {
                        if( JoinRequestTrials >= MaxJoinRequestTrials )
                        {
                            LoRaMacState &= ~LORAMAC_TX_RUNNING;
                        }
                        else
                        {
                            LoRaMacFlags.Bits.MacDone = 0;
                            // Sends the same frame again
                            OnTxDelayedTimerEvent( );
                        }
                    }
                }
                else
                {// Procedure for all other frames
                    if( ( ChannelsNbRepCounter >= LoRaMacParams.ChannelsNbRep ) || ( LoRaMacFlags.Bits.McpsInd == 1 ) )
                    {
                        if( LoRaMacFlags.Bits.McpsInd == 0 )
                        {   // Maximum repetitions without downlink. Reset MacCommandsBufferIndex. Increase ADR Ack counter.
                            // Only process the case when the MAC did not receive a downlink.
                            MacCommandsBufferIndex = 0;
                            AdrAckCounter++;
                        }

                        ChannelsNbRepCounter = 0;

                        if( IsUpLinkCounterFixed == false )
                        {
                            UpLinkCounter++;
                        }

                        LoRaMacState &= ~LORAMAC_TX_RUNNING;
                    }
                    else
                    {
                        LoRaMacFlags.Bits.MacDone = 0;
                        // Sends the same frame again
                        OnTxDelayedTimerEvent( );
                    }
                }
            }
        }

If my device is using classC and we receive a frame that is not intended for it between retransmissions, it will call  _PrepareRxDoneAbort()_ and set _LoRaMacFlags.Bits.McpsInd = 1_.
When _OnMacStateCheckTimerEvent_ is called it will increment "UpLinkCounter" and reset "ChannelsNbRepCounter " many times.

Is it a bug or I'm missing something ?

djaeckle, 2018-03-23:
Hi DoubleCheeseBacon,

thanks for your report. I will verify it as soon as possible.

djaeckle, 2018-08-01:
Hi DoubleCheeseBacon,

I see your point. I think its a bug as we should just ignore such a frame. I will set it on the roadmap for the next milestone.