Lora-net/LoRaMac-node #426: LoRa stack 4.4.0 SPI baud rate

ARAradtec, 2018-03-21:
Hello
sx1272 data sheet state that max SPI baud rate is 10Mhz
this is the function that calaculate the SPI prescalar:

void SpiFrequency( Spi_t *obj, uint32_t hz )
{
    uint32_t divisor = 0;
    uint32_t sysClkTmp = SystemCoreClock;
    uint32_t baudRate;
    while( sysClkTmp > hz )
    {
        divisor++;
        sysClkTmp = ( sysClkTmp >> 1 );

        if( divisor >= 7 )
        {
            break;
        }
    }
    baudRate =( ( ( divisor & 0x4 ) == 0 ) ? 0x0 : SPI_CR1_BR_2 ) |
              ( ( ( divisor & 0x2 ) == 0 ) ? 0x0 : SPI_CR1_BR_1 ) |
              ( ( ( divisor & 0x1 ) == 0 ) ? 0x0 : SPI_CR1_BR_0 );

    obj->Spi.Init.BaudRatePrescaler = baudRate;
}
the CR1 BR defines are:
#define SPI_CR1_BR_Pos              (3U)                                       
#define SPI_CR1_BR_Msk              (0x7U << SPI_CR1_BR_Pos)                   /*!< 0x00000038 */
#define SPI_CR1_BR                  SPI_CR1_BR_Msk                             /*!< BR[2:0] bits (Baud Rate Control) */
#define SPI_CR1_BR_0                (0x1U << SPI_CR1_BR_Pos)                   /*!< 0x00000008 */
#define SPI_CR1_BR_1                (0x2U << SPI_CR1_BR_Pos)                   /*!< 0x00000010 */
#define SPI_CR1_BR_2                (0x4U << SPI_CR1_BR_Pos)                   /*!< 0x00000020 */
this are the actual prescalers values:
Bits 5:3 BR[2:0]: Baud rate control
000: fPCLK/2
001: fPCLK/4
010: fPCLK/8
011: fPCLK/16
100: fPCLK/32
101: fPCLK/64
110: fPCLK/128
111: fPCLK/256

STM32 can handle SPI baud of systemclk/2
This code generate PCLK/8 (SPI_CR1_BR_1 is actualy /8) for 32Mhz
and PCLK/4 (SPI_CR0_BR_1 is actualy /8)  for 16Mhz

Is that correct?

BR
AR

ARAradtec, 2018-03-21:
hello again
Corrections for the last line:
and PCLK/4 (SPI_CR1_BR_0 is actually /4) for 16Mhz

DoubleCheeseBacon, 2018-03-22:
Hi,
I have made a change to have the good prescaler, for me _divisor_ is incremented once too much.

```
void SpiFrequency( Spi_t *obj, uint32_t hz )
{
    uint32_t divisor = 0;
    uint32_t sysClkTmp = SystemCoreClock;
    uint32_t baudRate;

    while( sysClkTmp > hz )
    {
        divisor++;
        sysClkTmp = ( sysClkTmp >> 1 );

        if( divisor >= 7 )
        {
            break;
        }
    }

    /* AJT -  according to the STM32 datasheet
    Bits 5:3 BR[2:0]: Baud rate control
    000: fPCLK/2
    001: fPCLK/4
    010: fPCLK/8
    011: fPCLK/16
    100: fPCLK/32
    101: fPCLK/64
    110: fPCLK/128
    111: fPCLK/256
    by default Spi CLK is divided at least by 2, so divisor need to be decreamented! */

//  When I decrement _divisor_ I get the exact frequency.
   if ( divisor > 0 )
    {
      divisor --;
    }

    baudRate =( ( ( divisor & 0x4 ) == 0 ) ? 0x0 : SPI_CR1_BR_2 ) |
              ( ( ( divisor & 0x2 ) == 0 ) ? 0x0 : SPI_CR1_BR_1 ) |
              ( ( ( divisor & 0x1 ) == 0 ) ? 0x0 : SPI_CR1_BR_0 );

    obj->Spi.Init.BaudRatePrescaler = baudRate;
}
```
My case : 
SystemCoreClock = 8MHz.
hz = 4MHz.
So I get divisor = 0.

ARAradtec, 2018-03-22:
OR you can start with system clock already divided by 2:

void SpiFrequency( Spi_t *obj, uint32_t hz )
{
    uint32_t divisor = 0;
    uint32_t sysClkTmp = SystemCoreClock >> 1;//prescaler value of 0 is already /2;
    uint32_t baudRate;

    while( sysClkTmp > hz )
    {
        divisor++;
        sysClkTmp = ( sysClkTmp >> 1 );

        if( divisor >= 7 )
        {
            break;
        }
    }

    /* AJT -  according to the STM32 datasheet
    Bits 5:3 BR[2:0]: Baud rate control
    000: fPCLK/2
    001: fPCLK/4
    010: fPCLK/8
    011: fPCLK/16
    100: fPCLK/32
    101: fPCLK/64
    110: fPCLK/128
    111: fPCLK/256
    by default Spi CLK is divided at least by 2! */

    baudRate =( ( ( divisor & 0x4 ) == 0 ) ? 0x0 : SPI_CR1_BR_2 ) |
              ( ( ( divisor & 0x2 ) == 0 ) ? 0x0 : SPI_CR1_BR_1 ) |
              ( ( ( divisor & 0x1 ) == 0 ) ? 0x0 : SPI_CR1_BR_0 );

    obj->Spi.Init.BaudRatePrescaler = baudRate;
}

DoubleCheeseBacon, 2018-03-22:
True

mluis1, 2018-07-02:
Can we close this issue?

mluis1, 2018-07-26:
No activity, we close the issues.

closed 2018-07-26 by mluis1