Lora-net/LoRaMac-node #429: Maximum Payload Lenght in US915

Adamzat, 2018-03-28:
Hi every one,

I am using the 4.4.0 stack, and I did many tests for US915 region. My device works very well, however, I have an issue regarding sending Data in the default data rate DR0 (SF10).

The device sends data of 10 bytes correctly. The Maximum allowed payload length for US in SF10 is 11 bytes (taking into account Fopts field). My issue, is that, if at any given time, the device has to respond to a MAC command (eg: >= 2 bytes, for a LinkAdrReq), the device goes into error.  The device exceeds the Max size of 11 bytes and therefore the device becomes unavailable.

Please, how can I fix it, I am thinking to omit data if the device has to respond to a Mac command? Is there a possibility to fix it in LoRaMacQueryTxPossible function? Any proposition?

Thank you for your help.

AdamZat

mluis1, 2018-07-02:
On the provided examples if the `LoRaMacQueryTxPossible` returns that the maximum size is exceeded then an empty uplink is generated in order to flush the MAC commands buffer.

closed 2018-07-02 by mluis1

utkarshshah007, 2019-01-22:
@mluis1 Ran into this issue recently - would clarify that in the example provided, there is no attempt to resend the original message after the empty uplink - it simply continues to the next scheduled uplink, since the empty uplink is still considered a successful send.

Shouldn't the `SendFrame( void )` method return false if a empty uplink had to be generated?

BenAtPip, 2019-01-22:
> @mluis1 Ran into this issue recently - would clarify that in the example provided, there is no attempt to resend the original message after the empty uplink - it simply continues to the next scheduled uplink, since the empty uplink is still considered a successful send.
> 
> Shouldn't the `SendFrame( void )` method return true (indicating failure) if a empty uplink had to be generated?

I agree on this, we had this exact issue under using a SF10 join on a network which disables the Dwell time following the first frame. If there was a change to indicate when a payload exceeded the allowed max payload size returning an error would be an improved implementation.