Lora-net/LoRaMac-node #440: AdrAckCounter doesn't increment for confirmed uplinks that don't get acknowledged

jamesl-dm, 2018-04-11:
I could be wrong, since my version of the stack diverged from upstream some time ago, but it looks like the AdrAckCounter is only incremented when an unconfirmed uplink fails to receive a downlink. When a confirmed uplink receives no downlink, it remains the same?

djaeckle, 2018-04-18:
Hi jamesl-dm,

which version are you using?

jamesl-dm, 2018-04-18:
I'm using a heavily modified 4.4.0. I had to add a line to it to make the counter increment for Confirmed Ups:

```
        if( ( AckTimeoutRetry == true ) && ( ( LoRaMacState & LORAMAC_TX_DELAYED ) == 0 ) )
        {// Retransmissions procedure for confirmed uplinks
            AckTimeoutRetry = false;
            if( ( AckTimeoutRetriesCounter < AckTimeoutRetries ) && ( AckTimeoutRetriesCounter <= MAX_ACK_RETRIES ) )
            {
                ...
            }
            else
            {
                RegionInitDefaults( LoRaMacRegion, INIT_TYPE_RESTORE );

                LoRaMacState &= ~LORAMAC_TX_RUNNING;

                MacCommandsBufferIndex = 0;

                NodeAckRequested = false;
                McpsConfirm.AckReceived = false;
                McpsConfirm.NbRetries = AckTimeoutRetriesCounter;
                if( IsUpLinkCounterFixed == false )
                {
                    UpLinkCounter++;
                    AdrAckCounter++;         // Added this line
                }
            }
        }
```

I can't see any other increments of AdrAckCounter for confirmed downlink failure in the current development branch, so I think you might need the change too.

djaeckle, 2018-08-01:
Hi jamesl-dm,

sorry for the late answer. Is your issue still present? We have restructured the code a little bit and I think this issue has been resolved, although a commit reference is missing. The MAC uses the function `StopRetransmission` for setting up the AdrAckCounter for both, confirmed and unconfirmed messages.

jamesl-dm, 2018-08-01:
Hi djaeckle. I haven't had a chance to check yet - thanks for letting me know.

closed 2018-08-01 by jamesl-dm