Lora-net/LoRaMac-node #441: AU915 / US915 RegionXX915NextChannel contains code to ensure strictly even use of 500 kHz channels, but doesn't use channels evenly

jamesl-dm, 2018-04-12:
The AU915 and US915 regions have a ChannelsMaskRemaining variable that counts off the used channels, so that all 64 + 8 of them are used once before any channel is reused. It works for the lower 64 channels, but doesn't work for the top 8 (500 kHz) channels.

In RegionXX915NextChannel there is code that specifically checks if the 500 kHz channels have all been used, and refreshes the ChannelsMaskRemaining, so counting them off one by one seems to be the intention:

```
// Check other channels
if( nextChanParams->Datarate >= DR_6 )
{
    if( ( ChannelsMaskRemaining[4] & 0x00FF ) == 0 )
    {
        ChannelsMaskRemaining[4] = ChannelsMask[4];
    }
}
```

But RegionCommonChanDisable is called with the max channels reduced, so the 500 kHz channels are never counted off after use:

`RegionCommonChanDisable( ChannelsMaskRemaining, *channel, AU915_MAX_NB_CHANNELS - 8 );`

mluis1, 2018-04-12:
The FCC only mandates to use randomly all the 125 kHz before restarting the procedure. 
For the 500 kHz channels there is no such constraint. The channels can be used without any constraint.

This means that the current implementation is compliant with FCC regulations.

jamesl-dm, 2018-04-12:
I'm not commenting on FCC compliance - I'm saying the code is inconsistent.