Lora-net/LoRaMac-node #443: Class C need help

vkiselyov, 2018-04-16:
Hi.
I try to launch Class C example.
I build project and write it to RAK811. Some my devices send messages and I see it on getaway. I would like this messages on Class C device, but I don't see it on device.
Method OnRadioRxDone never called.

What I should to do for receive messages on my Class C device?

djaeckle, 2018-06-21:
Hi vkiselyov,

please refer to the example implementations for class c. The class c feature is available by default. To switch to class c, you need to perform an MIB request after MAC initialization:

```
MibRequestConfirm_t mibReq;
mibReq.Type = MIB_DEVICE_CLASS;
mibReq.Param.Class = CLASS_C;
LoRaMacMibSetRequestConfirm( &mibReq );
```

nestorayuso, 2018-06-21:
Acording to the specification, a device does not report to the network server that it is class-c or changed to class-c.

You must indicate in the network server that the device is class-c.

mluis1, 2018-07-26:
No activity, we close the issue.

closed 2018-07-26 by mluis1