Lora-net/LoRaMac-node #445: ADC Enabling Issue on STM32 boards

temporaryaccount, 2018-04-18:
adc-board.c AdcMcuReadChannel function uses HAL's low level __HAL_ADC_ENABLE macro function to enable the ADC and follows to immediately use it. I was having problems, after some debugging I found out that adding one millisecond wait after enabling the ADC solves my issue:

```C
    // Enable ADC1
    __HAL_ADC_ENABLE( &AdcHandle );

    // Added by me
    HAL_Delay(1);

    // Start ADC Software Conversion
    HAL_ADC_Start( &AdcHandle );

    HAL_ADC_PollForConversion( &AdcHandle, HAL_MAX_DELAY );
```

After more investigation I found out in STM32L0 reference manual that we must wait for ADRDY bit to become 1 before we can use the ADC:

    Bit 0 ADEN: ADC enable command

    This bit is set by software to enable the ADC. **The ADC will be effectively ready to operate once the
    ADRDY flag has been set.**

    It is cleared by hardware when the ADC is disabled, after the execution of the ADDIS command.
    0: ADC is disabled (OFF state)
    1: Write 1 to enable the ADC.

    Note: Software is allowed to set ADEN only when all bits of ADC_CR registers are 0 (ADCAL=0,
    ADSTP=0, ADSTART=0, ADDIS=0 and ADEN=0)

HAL's own internal ADC_Enable function is doing it this way:

```C
    /* Enable the ADC peripheral */
    __HAL_ADC_ENABLE(hadc);
    
    /* Delay for ADC stabilization time. */
    ADC_DelayMicroSecond(ADC_STAB_DELAY_US);

    /* Get tick count */
    tickstart = HAL_GetTick();
    
    /* Wait for ADC effectively enabled */
    while(__HAL_ADC_GET_FLAG(hadc, ADC_FLAG_RDY) == RESET)
    {
      if((HAL_GetTick() - tickstart) > ADC_ENABLE_TIMEOUT)
      {
        /* Update ADC state machine to error */
        SET_BIT(hadc->State, HAL_ADC_STATE_ERROR_INTERNAL);
        
        /* Set ADC error code to ADC IP internal error */
        SET_BIT(hadc->ErrorCode, HAL_ADC_ERROR_INTERNAL);
        
        return HAL_ERROR;
      }
    }
```

I suggest either to use HAL's higher level functions or to implement the procedure correctly.

temporaryaccount, 2018-04-18:
Here is my final working version of the `AdcMcuReadChannel` function:

```C

uint16_t AdcMcuReadChannel( Adc_t *obj, uint32_t channel )
{
    ADC_ChannelConfTypeDef adcConf = { 0 };
    uint16_t adcData = 0;
    uint32_t tickStart = 0;
    bool isAdcReady = true;

    // Enable HSI
    __HAL_RCC_HSI_ENABLE( );

    // Wait till HSI is ready
    while( __HAL_RCC_GET_FLAG( RCC_FLAG_HSIRDY ) == RESET )
    {
    }

    __HAL_RCC_ADC1_CLK_ENABLE( );

    adcConf.Channel = channel;
    adcConf.Rank = ADC_RANK_CHANNEL_NUMBER;

    HAL_ADC_ConfigChannel( &AdcHandle, &adcConf );

    // Enable ADC1
    __HAL_ADC_ENABLE( &AdcHandle );

    // Wait for ADC effectively enabled
    tickStart = HAL_GetTick( );
    while ( __HAL_ADC_GET_FLAG( &AdcHandle, ADC_FLAG_RDY ) == RESET )
    {
        if ( ( HAL_GetTick( ) - tickStart ) > ADC_ENABLE_TIMEOUT )
        {
            isAdcReady = false;
            break;
        }
    }

    if ( isAdcReady )
    {
        // Start ADC Software Conversion
        HAL_ADC_Start( &AdcHandle );

        HAL_ADC_PollForConversion( &AdcHandle, HAL_MAX_DELAY );

        adcData = HAL_ADC_GetValue( &AdcHandle );
    }

    __HAL_ADC_DISABLE( &AdcHandle );

    if( ( adcConf.Channel == ADC_CHANNEL_TEMPSENSOR ) || ( adcConf.Channel == ADC_CHANNEL_VREFINT ) )
    {
        HAL_ADC_DeInit( &AdcHandle );
    }
    __HAL_RCC_ADC1_CLK_DISABLE( );

    // Disable HSI
    __HAL_RCC_HSI_DISABLE( );

    return adcData;
}

```

mluis1, 2019-02-12:
As there is no activity for quite long time we close the issue.

Regimbal, 2019-04-03:
Sorry to ask, but I had the same issue with B-L072Z-LRAWAN1 (STM32L072).
Why not just get rid of __HAL_ADC_ENABLE and let HAL_ADC_START handle it like it is suggested in STM32 example codes?
That solves the issue for me.

mluis1, 2019-04-03:
Please find attached a patch file implementing @temporaryaccount proposition for each supported platform.

We haven't tested these changes on the different platforms.

[adc-read-fix.patch.txt](https://github.com/Lora-net/LoRaMac-node/files/3037776/adc-read-fix.patch.txt)

Regimbal, 2019-04-04:
Seems to work on my B-L072Z-LRWAN1

Regimbal, 2019-04-12:
By the way, patch is not complete. It lacks some #include <stdbool.h> in adc-board.c

mluis1, 2019-04-18:
Re-opened issue in order to not forget to apply the patch

selimsagir, 2019-04-24:
> Sorry to ask, but I had the same issue with B-L072Z-LRAWAN1 (STM32L072).
> Why not just get rid of __HAL_ADC_ENABLE and let HAL_ADC_START handle it like it is suggested in STM32 example codes?
> That solves the issue for me.

Hi, how you 

> Seems to work on my B-L072Z-LRWAN1

Hi Regimbal, how you use source code with B-L072Z-LRWAN1. I tried  import  code to  TrueStudio project which was created through CubeMX , however I am failed again and again.

closed 2019-06-13 by mluis1

MarkHimself, 2021-11-17:
I have an STM32G031K8 board and have also experienced issues with the ADC enable procedure.
Specifically, with the ADC_ADRDY bit. This is a status bit that turns on when the ADC is ready (after you set the ADEN bit).

To explain the problem I have, I will provide a quick procedure.
1.  I setup the ADC after the board is powered up (enable the ADC clock, voltage regulator, calibration, etc)
2.  I enable the ADC for the first time (following the procedure in the manual for my board)
2.a. Clear the ADRDY bit
2.b. Enable the ADC by setting the ADEN bit
2.c. Wait for the ADRDY bit in the status register to turn to wait. It turns "1" this time.
3. I disable the ADC by setting the ADDIS bit. (I've tried clearing the ADRDY bit and not clearing it as the manual says it's optional)
4. I enable the ADC by setting the ADEN bit.
5. The ADRDY bit NEVER turns on.

In short, the issue is that the ADRDY bit turns on the first time I enable the ADC but if I disable the ADC and then enable it again, the ADRDY bit will not turn on. I've tracked this with a while() loop to wait for the ADRDY bit to turn on and with interrupts. It never turns on the second time. However, the ADC is still enabled and functional.

I would be interested for someone to repeat the steps listed above and see what they get.

Here is some parts of my code (enough to get the general idea of what is going on and replicate with a different board)

```

void ADC_Example1(){
	// ADC control
	Enable_ADC_Clock();
	ADC_Disable();
	ADC_ClockMode2();
	ADC_EnableVoltageRegulator();
	ADC_Calibration();
	
	ADC_Enable();                                             // First time ADC is enabled
	
	ADC_DiscontinuousMode();
	ADC_12bit_resolution();
	ADC_EnableAutoOffMode();
	ADC_ChannelInputMode();
	
	ADC_ChannelSelection(10);
	
	while(1){
		ADC_StartConversion();
		ADC_WaitForEndOfConversion();
		adc_data = ADC_getData();
		delay_s(1);
		
		ADC_Disable();
		ADC_ClockMode2();
		ADC_EnableVoltageRegulator();
		ADC_Calibration();
		ADC_Enable();                                  // Second time ADC is enabled.
	}
}

void Enable_ADC_Clock(){
	RCC->APBENR2 |= RCC_APBENR2_ADCEN;					// Enable clock to ADC		pg. 201
}

// procedure is outlined on pg. 349
void ADC_Disable(){
	uint32_t temp = 0;
	
	temp = ADC1->CR;
	
	if (!(temp & ADC_CR_ADEN)) return;
	
	if (temp & ADC_CR_ADSTART){			// if there is an ongoing conversion
		ADC1->CR = ADC_CR_ADSTP;			// stop the conversion
		while(ADC1->CR & ADC_CR_ADSTP);	// wait until the conversion is stopped
	}
	
	ADC1->CR = ADC_CR_ADDIS;				// disable the ADC
	while(ADC1->CR & ADC_CR_ADEN);		// wait until the ADC is disabled
	while(ADC1->CR & ADC_CR_ADDIS);
	
	//ADC1->ISR = ADC_ISR_ADRDY;			// Clear the ADC Ready bit in the status register
}

// procedure is outlined on pg. 349
void ADC_Enable(){
	
	if (ADC1->CR & ADC_CR_ADEN) return;	// if the ADC is already enabled
	
	ADC1->ISR = ADC_ISR_ADRDY;			// Clear the ADC Ready bit in the status register
	ADC1->CR = ADC_CR_ADEN;
	delay_ms(1);                                                    // I've replaced the below while() loop with 1 millisecond delay
	//while(!(ADC1->ISR & ADC_ISR_ADRDY));	// wait while it's not ready (stops waiting when it's ready) - the ADRDY bit is not reliable
}

```