Lora-net/LoRaMac-node #446: Multicast feature

marianodo, 2018-04-19:
Hi everyone
Could any one test Multicast feature? I added a new MulticastChannel so I can detect when a multicast message is for my node but when the library checks the function "LoRaMacComputeMic" (I supposes that function checks "mic" with the encryption key) I always get false, so then the message is discarded.

I need some help
Thanks

nestorayuso, 2018-04-19:
What vloramac version/branch are you using? what Region?
Are you using Class-C or Class-B?

you need to add a multicast address to the device and multicast-NwkSKey and multicast-AppSKey and DownLinkCounter.

basically you must fill one or more MulticastParams_t structures with the fuction LoRaMacMulticastChannelLink()

marianodo, 2018-04-19:
I'm using AU915 in Class-C
I've already add a multicastchannel struct with LoRaMacMulticastChannelLink()
I  can't find version/branch of my code. I clone this repo months ago.

mluis1, 2018-07-02:
Personally, I have tested the multicast using a US915 ClassB device and everything worked as expected. For ClassC it should also work as it is simpler than for the ClassB.

Have you solved your issue? Can we close the issue?

Please find below some hints on how to solve your issue:

* Does your network server support multicast channels?
* Have you created the multicast channel on the network server?
* Are the multicast keys and the address the same on both sides?

mluis1, 2018-07-26:
No activity, we close the issue.

closed 2018-07-26 by mluis1

cocolitto, 2022-05-10:
@mluis1 

Hello and big thanks for your work regarding FUOTA.

At the moment I'm trying to set up a FUOTA session at my labor enviroment at the university,

I'm using TheThingsStack network server and this FUOTA server implementation: 
[https://github.com/elsalahy/test-fuota-server](url)

For the node the LoRaMAC-node FUOTA example with the mac version 1.1.0 is used.

All the published example for the ClockSync, Multicast and fragmented data block transport are using the mac version 1.0.3.

The clock sync and fragmented data block trasnport works fine but the multicast session doesn't work because the node falls in a mic_fail error when receiving a multicast downlink. I think its because of wrong keys provided. So it receives the downlink and the address is correct but the calculation of the mic fails.

Could you give me a example key set-up for the se-identity.h file for a multicast session and how the multicast device on the network server has to be configured? 
Or maybe list some keypoint what keys I have to set up for the lorawan mac version 1.1.0 on the node and server for a multicast session?

Best regards Corsin Obrist

mluis1, 2022-05-13:
@cocolitto Could you please ask your question under [Discussions](https://github.com/Lora-net/LoRaMac-node/discussions) tab?

It is the usage to not hijack closed issues/topics which are not related with the new question.
Usually we do not answer closed topics.

Thanks in advance.