Lora-net/LoRaMac-node #449: RSSI stability with two LoRaMac versions

Lipnel, 2018-04-26:
Hello,

I have two basic applicative softwares, one running with the LoRaMac stack v3.1.0 and one running with the LoRaMac stack v4.3.2. The application test that I run is design to send periodically an uplink with fix data payload on SF12 with a txpower of 14dBm.

Here is my issue, on the application with old LoRaMac stack (v3.1.0) I observe that the uplink RSSI received by the gateway is quite stable with an average delta of ±5 dBm.
With the application with new LoRaMac stack (v4.3.2) I observe that the uplink RSSI received by the gateway is not at all stable with a period of 1 hour of stable rssi values then 1 hour where the delta value between two messages can be  ±20 dBm.

Does anyone have experienced that ?

Thanks for your help.
Regards

Lipnel, 2018-05-17:
Hello,
Here is a screenshot of my device with stack 4.x, it's the RSSI based on a time period of 7hours. The data payload and the uplink frequency never changes. We can see the RSSI oscillate with a periodic form.

[image]

With the same hardware, same payload and same uplink frequency, I have RSSI values as we can see below with the stack 3.x. We can see there is no such fluctuation rate.

[image]

mluis1, 2018-07-02:
Can we close this issue?

closed 2018-07-03 by Lipnel