Lora-net/LoRaMac-node #453: Problem via OTAA

Lozeonjyu, 2018-05-03:
Hi guys,

I am currently using sx1261 shield on top of Nucleo LR073RZ. The data is showing up using ABP method but not OTAA, looks like there is a registration problem.

Dev EUI, App key is correct. But the TX light doesn't blink on sx1261 board.

Did I miss some setups?

Many thanks

ponyatov, 2018-05-14:
Please check did you mix AppKey (OTAA) vs AppSKey (ABP) -- I did this typo-like error and lost many many days with debugging of strange join behavior.

Lozeonjyu, 2018-05-16:
@ponyatov Thanks for the mention. However, this is not the problem in my case after I double check them. But I do appreciate your help

mluis1, 2018-07-02:
Can we close this issue?

mluis1, 2018-07-26:
No activity, we close the issue.

closed 2018-07-26 by mluis1