Lora-net/LoRaMac-node #458: ADR_ACK_LIMIT and ADR_ACK__DELAY should be valiable.

ty4tw, 2018-05-09:
ADR_ACK_LIMIT and ADR_ACK_DELAY are constant values define by #define statement.
Those values depends on the transmission interval.
If the end device sends a packet  once a hour,  It can not be confirmed until after 3 days.
It' too late for my application.

nestorayuso, 2018-05-09:
ADR_ACK_LIMIT and ADR_ACK_DELAY are fixed as of LoRaWAN 1.0.2 specification.

In the upcoming implementation, ADR_ACK_LIMIT and ADR_ACK_DELAY will be variable as of LoRaWAN 1.1 specification.

Note that ADR_ACK_LIMIT and ADR_ACK_DELAY are used for the ADR algorithm, not for ACK the uplinks.

The node will set the ADR_ACK bit after 64 unACKed uplinks. And after another 32 unACKed uplinks, the node will lower the datarate itself. This can be last up to 3 days in your case, but the network server is free to send an ADR_REQ earlier, not only when the node asks for it with the ADR_ACK bit.

The problem with LoRaWAN 1.0.2 is that the ADR algorithm needs at least 1 downlink every 64+32 uplinks and this is may be lot in some regions or in congested networks. So ADR_ACK_LIMIT and ADR_ACK_DELAY are supposed to be even bigger in LoRaWAN 1.1

Your device packet will be confirmed immediately using RX1 or RX2 windows. If your gateway has big latency because using GPRS... your device packet will be confirmed in an hour using the next RX1 or RX2 windows.

But remember, in LoRaWAN it is not a good practice to use confirmed uplinks everytime, use it just for critical messages.

ty4tw, 2018-05-09:
Thank you for your quick reply.
I will implement own AdrNext() function in my application to avoid frequently confirm packet.

closed 2018-05-09 by ty4tw