Lora-net/LoRaMac-node #463: some rxErrors running ping-pong program with REGION_CN470 and USE_MODEM_LORA on

YJGit, 2018-05-15:
I run the ping-pong program with REGION_CN470 and  USE_MODEM_LORA on. Two nodes are about two meters apart, which uses 5dbi antenna。I find that there would be 2 or 5 or even worse 20 rxErrors in about 1500 ping-pong packets, and the number of errors were not uniformly distributed in two nodes. I read the codes, finally I find follow codes in sx1276.c:
    case MODEM_LORA:
                {
                    int8_t snr = 0;
                    // Clear Irq
                    SX1276Write( REG_LR_IRQFLAGS, RFLR_IRQFLAGS_RXDONE );
                    irqFlags = SX1276Read( REG_LR_IRQFLAGS );

when rxerror happens, irqFlags would be 32, which means PAYLOADCRCERROR. I think the numbers of errors are a bit more, but I don't know what to do next. Can you give me some advice?

GregCris, 2018-07-03:
If your nodes are in close proximity with 5dBi antenna, you are probably saturating the receiver. Try to decrease the output power or to move the nodes car one from the other.

mluis1, 2018-07-26:
Can we close the issue?

YJGit, 2018-08-08:
Yes.

closed 2019-02-12 by mluis1