Lora-net/LoRaMac-node #464: Node doesn't go to sleep when LoRaMacMlmeRequest doesn't return LORAMAC_STATUS_OK

XKottelaar, 2018-05-16:
Hello,

I found out due to the implementation of issue #296 that the node doesn't go to sleep when the 'LoRaMacMlmeRequest()' for a OTAA join doesn't return 'LORAMAC_STATUS_OK'. When ok isn't returned the node will go to the 'DEVICE_STATE_CYCLE' in which the 'TxNextPacketTimer' is set. However the 'TxDutyCycleTime' to which the timer will be set is uninitialized at this point. The 'TxDutyCycleTime' is only set when the device has been in the 'DEVICE_STATE_SEND' state, as far as I could find. 

This will set the timer to '0' which results to the uC not sleeping until OK is returned.

I noticed this issue because the MAC now also returns a NOK value when it there is no channel available to transmit the message due to duty cycle limitations (#296). This happens when an SF of 8 or higher is needed in order to join the network.

alexanderplanar, 2018-05-28:
Yep, I confirm the issue. I'd got the same problem last weekend. The initializing the variable fixes the problem.

XKottelaar, 2018-07-17:
Looking at the code, it would indeed fix the issue.

mluis1, 2018-07-17:
Thanks for the feedback. I then close the issue.

closed 2018-07-17 by mluis1