Lora-net/LoRaMac-node #465: MAC and ACK skipped because of Dutycycle ?

DoubleCheeseBacon, 2018-05-17:
Hi everybody,

If the Gateway send a confirmed downlink with MAC Request, the end device should answer by a MAC answer and the ACK bit.

In "PrepareFrame" we add the ACK bit and the MAC anwser but if the frame is not sent because of dutycycle limitation we will lost the anwser and the ACK bit.
In the next uplink frame they won't be added because "MacCommandsInNextTx" and "SrvAckRequested" are "false".

Is it a bug or I'm missing something ?

Thanks.

djaeckle, 2018-06-21:
Hi DoubleCheeseBacon,

sorry for the late response and thanks for your report. We can confirm that this is an issue. As we are going to update the project soon, we will verify that this issue is fixed.

mluis1, 2018-07-02:
There is no specific commit for this issue resolution.

Could you please check that with the commit c9d4f25 you don't see the issue anymore?

mluis1, 2018-07-26:
No activity, we close the issue.

closed 2018-07-26 by mluis1

MaheshKumarPN, 2018-09-21:
Do we have any solution or fix for in stack version v4.4.1?
I am also facing the same issue