Lora-net/LoRaMac-node #466: File Does Not Exist!

1405224, 2018-05-18:
Hi,
I have followed the instructions to implement this project and now I'm facing the errors below:
First when i build, I get:
[build] make[2]: *** [src/apps/LoRaMac/LoRaMac-classA] Error 1
[build] make[1]: *** [src/apps/LoRaMac/CMakeFiles/LoRaMac-classA.dir/all] Error 2
[build] make: *** [all] Error 2
[build] Build finished with exit code 2

Then, when i try to Debug, i get a popup message:
launch: program '/Users/LoRaMac-node/build/src/apps/LoRaMac/LoRaMac-classA' does not exist

Also, in the 'Problems' tap I've got:
Property MIDebuggerPath is not allowed

Could anyone please help me??

Thanks

mluis1, 2018-07-02:
Can we close this issue?

mluis1, 2018-07-26:
No activity, we close the issue.

closed 2018-07-26 by mluis1