Lora-net/LoRaMac-node #469: Compiling us915 - ld returns 5

lemonbuzz, 2018-05-23:
So I have little to no experience with embeded systems, I have played around with Arduino's but never got my hands dirty with something like an STM32 so I am struggling getting everything set up. Right now, I have the STM32L073RZHx. This board also has a LoRa radio and the manufacturer (Acsip) has provided me with working example projects in the Keil IDE. I don't have any issues compling and running those but for the scope of the project, we cannot justify to buy the Keil IDE once the 7 day trial period ends.

I heard about free alternatives, gave Atollic TrueSTUDIO a try. Obviously, I could not import the Keil projects directly into TrueSTUDIO, so i tried to rebuild a project structure, step-by-step, incrementally adding all the dependencies and solving whatever error would show. However, I have now hit a wall and i cannot seem to solve my latest issue. When I try to compile, 'ld' stops working and returns 5. I know ld is the linker but i have no idea what is causing this or where to look for. Reading for help online just makes everything a lot more confusing to me and my time is ticking! Help is very very appreciated.

Thank you.

(please ask for more info if needed)

mluis1, 2018-07-03:
Can we close this issue?

lemonbuzz, 2018-07-05:
Yes. Funny thing, Keil is free for L0 boards, but I had to ask for it. I'm all good. Thanks

closed 2018-07-06 by mluis1