Lora-net/LoRaMac-node #470: What's next

1405224, 2018-05-25:
I have followed the instruction and completed the implementation, 
the code is debugged and running but how can i see whats happening? 
I meant i tried to add the device to the things network but i cant see and data being received by the gateway? 
I have configured the Commissioning.h file to match with my TTN and gateway.

Please advise?

Thanks

ZuratonE, 2018-05-29:
Have you you configured your gateway? Is it visible in gateways list?
What kind of activation do you use?
Have you added your device's eui's ant etc to thingnetwork?

1405224, 2018-05-29:
Hi ZuratonE,
Thanks for your reply, 
I have now managed to connect the device with the gateway and I can see the data in TTN using class A.
However, when i try to use class B, nothing shows up in TTN. is that because TTN does not support class B yet? if so, how can see the data being sent/received at the gateway? or to chack that class B does work?

Really appreciate your help..
Thanks

1405224, 2018-05-30:
Hi again ZuratonE,
When I connect my device to TTN using class B I can see packets being received by the gateway, however, in the device page (data tap) i can't see any data being sent?
Also is there a better way to monitor the data flow using class B? i.e. to monitor not only data packets but beacons and ping slots etc?

Thanks

mluis1, 2018-07-03:
Nowadays TTN network server doesn't support ClassB or ClassC devices.

The support for these classes will be added to TTN for their next release V3 if I'm not mistaken.

closed 2018-07-03 by mluis1