Lora-net/LoRaMac-node #473: the total millisecond 's overflow problem

victor888866, 2018-05-28:
I have some questions about Lora-net/LoRaMac-node/branch/class_b

It will be much appreciate if I can get help from you

1、I use apps\LoRaMac\classB\NucleoL073

2、the code “TimerTime_t currentTime = TimerGetCurrentTime( );”

the fuction is getting the total millisecond ？
but TimerTime_t is uint32_t
so it has a overflow problem

for example

sysTime.Seconds=1527471204
sysTime.SubSeconds=469
so the currentTime will be (1527471204 *1000+469) & 0xFFFFFFFF=0xA460E875

but in LoRaMacClassBDeviceTimeAns fuction because of the total millisecond 's overflow problem

for example
when currentTime %128000=0
but systime(millisecond ) % 128 may ≠0

while the beacon will send at the time witch modulo 128 equal 0，so if I set the beacon RXwindow by “currentTime ”，the sysTime will diffrent from the gateway send the beacon

best regards !

djaeckle, 2018-06-21:
Hi victor888866,

i'm not sure if i understand you question correctly. Would you please reprahse it. Thanks.

victor888866, 2018-06-21:
@djaeckle  hi djaeckle
for example
If now utc is 1529624648，the code “TimerTime_t currentTime = TimerGetCurrentTime( );
does it return 1529624648-315964800=1213659848
or currentTime  is just a relative value？

mluis1, 2018-06-22:
We will soon update the develop branch. We are finalizing some basic tests.

For this update we have completely refactored the RTC drivers, timer.c/h and we have added a systime.c/h module implementation with ST help.

We believe that with the new implementation it will be easier to understand how things are done.

closed 2018-07-03 by mluis1