Lora-net/LoRaMac-node #475: SNR value in DevStatusAns

branek, 2018-06-01:
Hi all,

I am aware of previous topics (#143, #104, #75 and #29) regarding this but I think that there is still error in reporting DevStatusAns.

In LoRaWAN 1.0.2, Chapter 5.5 (and same in 1.1) there is a clear explanation how to format of DevStatusAns answer:

[image]

In sx1276-7-8-9 datasheet, SNR register is explained as:

[image]

So, in my opinion, i think that sx1276.c driver should call RxDone with calculated value of SNR (lines from 1546 - 1557 on develop branch)), not register value.

Also, in current code LoRaMAC.c, dev answer bits 7-6 are masked, so we lost sign and it's not possible to calculate snr on application side now.

I'm willing to provide PR if you agree.

Thanks
Branek

mluis1, 2018-07-03:
I had a lot of discussions about this with the Alliance technical committee chair and the outcome is that the implementation is correct as is.

I do agree that this is quite confusing.

closed 2018-07-03 by mluis1

carloshvp, 2018-11-09:
Hello @mluis1  ,

I have discussed about this topic with several developers from other companies, we all agree that the explanation from @branek is correct and the current code has a bug.

There are even more PRs apart from those mentioned by @branek , trying to fix the same issue, for example: #323 

Would it be possible to give a reasonable explanation to the issue? Otherwise more and more developers will continue insisting on this topic. Maybe the datasheet has an errata?

Thank you in advance.
Carlos