Lora-net/LoRaMac-node #478: blocked on join following RX2_TIMEOUT

Regimbal, 2018-06-05:
Hello everyone,
We set our node messages to be acknowledged.
In the case the node does not receive its ACK, we wanted for some reason to rejoin the lora network.

Based on code in the develop branch, we appended in main.c / McpsConfirm the code as below:

> static void McpsConfirm( McpsConfirm_t *mcpsConfirm )
>     MuxLockFifoEvent = false;
>     if( mcpsConfirm->Status == LORAMAC_EVENT_INFO_STATUS_OK )
>     {
>     [...]
>     }
> } else if (mcpsConfirm->Status == LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT) {
>     	tryToJoinOTAA(); // code to join network OTAA
> 	}
> }

For some reason, the first time it works, that is to say our node successfully rejoins the network.
But starting the second time it does not receives its ACK, the join procedure does not end, and we get a Busy status.
This implies that as long as a rejoin and all others following fail, then our network is unable to send messages anymore.

Rejoining the network is probably not the best idea, but that is not the point. I'd like to understand it works a first time and not the others.

Thanks a lot,
Nicolas

nestorayuso, 2018-06-05:
I think if you send a confirmed message with NbTrials = 5, you need to wait for five LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT Mcps Confirmations to grant for a dead link.

Regimbal, 2018-06-05:
This is what I though in a first place... But I tested on an oscilloscope and in debug mode, and checked the consumption cycles.
It seems McpsConfirmation ends with LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT after the last try... not at each try.

nestorayuso, 2018-06-05:
I asked for documentation of the sequence of confirmations/indications in https://github.com/Lora-net/LoRaMac-node/issues/327

Now that the development of the stack is stalled until the v4.4.1 certification pass in test houses, I think it is a good time to document it, @djaeckle.

Regimbal, 2018-06-05:
Definitely, I'd already seen your issue in #327 and was quite intrigued and left without answer... which led me to post an issue.

mluis1, 2018-07-03:
When sending confirmed messages and the ACK is not received the MAC layer will autonomously silently re-transmit the same frame. The upper layer gets only notified at the end of the process. 

Inside the `McpsConfirm` function one can check the `NbTrials` field of McpsConfirm_t structure to know how many times the frame has been re-transmitted.

Rejoining the network is really not a good idea. If your session parameters haven't changed there is no reason to re-join the network.

closed 2018-07-03 by mluis1

ly1243667342, 2018-09-27:
@Regimbal  i have same issue with you , how do you solve this issue?