Lora-net/LoRaMac-node #483: Antenna gain not applied in US915

jamesl-dm, 2018-06-13:
RegionUS915TxConfig doesn't apply the max EIRP or antenna gain passed to it, like the other regions do. Is US915 a special case for transmit power handling, or is this just an omission?

mluis1, 2018-07-03:
Yes, US915 region is a special case.

Please refer to [LoRaWAN Regional Parameters v1.1rB](https://lora-alliance.org/resource-hub/lorawantm-regional-parameters-v11rb) document. 

LoRaWAN Regional Parameters v1.0.3rA is not yet officially released on the alliance website. But both documents are almost identical.

closed 2018-07-03 by mluis1

mluis1, 2018-07-03:
I understand your point but, for this project we must follow the LoRa-Alliance specifications and those currently say that for US915 we must not apply the antenna gain.

My advise is that you make a LoRaWAN specification change request. If the change request is accepted by the LoRa-Alliance and the specifications are updated accordingly then we will apply those changes to the project.

Please note that nothing prevents you to implement your proposition the way you describe it. The only constraint in the end is that you must be FCC compliant.

jamesl-dm, 2018-07-03:
_Where_ do they say that we must not apply the antenna gain? I want to comply with the specification too...

mluis1, 2018-07-03:
Have you read the LoRaWAN Regional Parameters specification?

For US915 region Table 13:
[image]

jamesl-dm, 2018-07-03:
Ah! I see. Thanks!