Lora-net/LoRaMac-node #490: MoteII platform removal

mluis1, 2018-06-19:
The MCU used on this platform doesn't have enough FLASH/RAM memory required to handle the ClassB.

closed 2018-07-02 by mluis1

temporaryaccount, 2018-07-04:
Sorry I'm late to the party, but did we really have to remove this platform's support.

We are successfully using STM32L051 for Class A solutions, and now it's gone.

temporaryaccount, 2018-07-04:
Maybe we could introduce some kind of "capabilities" instead? I.e. this platform supports Classes A, C, and that platform supports Classes A, C, etc.

mluis1, 2018-07-04:
Actually with latest commits done on develop branch ClassA and ClassC wont compile as well.

The STM32L051has 64K bytes of FLASH and 8K bytes of RAM memory. As you can see in the below compiling result with all optimizations the FLASH memory is overflowed by 8784 bytes.

This is mainly due to the addition of the software emulated secure-element. We will edit the original message to reflect this as well.

The STM32L051
```bash
[build] [100%] Linking C executable LoRaMac-classA
[build] c:/progra~2/gnutoo~1/72017-~1/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: LoRaMac-classA section `.text' will not fit in region `FLASH'
[build] c:/progra~2/gnutoo~1/72017-~1/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 8784 bytes
[build] collect2.exe: error: ld returned 1 exit status
[build] mingw32-make.exe[2]: *** [src\apps\LoRaMac\CMakeFiles\LoRaMac-classA.dir\build.make:331: src/apps/LoRaMac/LoRaMac-classA] Error 1
[build] mingw32-make.exe[1]: *** [CMakeFiles\Makefile2:397: src/apps/LoRaMac/CMakeFiles/LoRaMac-classA.dir/all] Error 2
[build] mingw32-make.exe: *** [Makefile:83: all] Error 2
[build] Build finished with exit code 2
```

temporaryaccount, 2018-07-04:
Thanks for the answer. I just reviewed the stm32l0.cmake file, we aren't using any kind of optimization.

If the removal decision isn't final, can you please try to compile with -Os (to optimize for size) instead of -Og, and also without the debug information (-g) and see what happens? Thanks!

mluis1, 2018-07-04:
**The compilation result previously shown is with `-Os` and without `-g`.**

The `stm32l0.cmake` is included by `toolchain-arm-none-eabi.cmake` which  will override the optimization flags depending on how the `cmake` utility is called.

if `cmake` is called with parameter `--config` set to `Release` then the `CMAKE_C_FLAGS_RELEASE`, `CMAKE_CXX_FLAGS_RELEASE`, `CMAKE_ASM_FLAGS_RELEASE`, `CMAKE_EXE_LINKER_FLAGS_RELEASE` will be used.

```cmake
# Options for RELEASE build
# -Os Optimize for size. -Os enables all -O2 optimizations
# -flto runs the standard link-time optimizer
set(CMAKE_C_FLAGS_RELEASE "-Os -flto" CACHE INTERNAL "C Compiler options for release build type")
set(CMAKE_CXX_FLAGS_RELEASE "-Os -flto" CACHE INTERNAL "C++ Compiler options for release build type")
set(CMAKE_ASM_FLAGS_RELEASE "" CACHE INTERNAL "ASM Compiler options for release build type")
set(CMAKE_EXE_LINKER_FLAGS_RELEASE "-flto" CACHE INTERNAL "Linker options for release build type")
```

if `cmake` is called with parameter `--config` set to `Debug` then the `CMAKE_C_FLAGS_DEBUG`, `CMAKE_CXX_FLAGS_DEBUG`, `CMAKE_ASM_FLAGS_DEBUG`, `CMAKE_EXE_LINKER_FLAGS_DEBUG` will be used.

```cmake
# Options for DEBUG build
# -Og enables optimizations that do not interfere with debugging
# -g produce debugging information in the operating system’s native format
set(CMAKE_C_FLAGS_DEBUG "-Og -g" CACHE INTERNAL "C Compiler options for debug build type")
set(CMAKE_CXX_FLAGS_DEBUG "-Og -g" CACHE INTERNAL "C++ Compiler options for debug build type")
set(CMAKE_ASM_FLAGS_DEBUG "-g" CACHE INTERNAL "ASM Compiler options for debug build type")
set(CMAKE_EXE_LINKER_FLAGS_DEBUG "" CACHE INTERNAL "Linker options for debug build type")
```

temporaryaccount, 2018-07-04:
OK then :(

Thanks again, and sorry for the trouble.

I guess we'll have to migrate to STM32L071 with 192KB of Flash and 20KB of RAM.

mluis1, 2018-07-04:
One more thing. 

If an hardware secure-element is used then the overall size of the firmware image could be reduced.

Unfortunately, I can't tell you by how much the size will be reduced as it will depend on the secure-element manufacturer choices. Please take a look at the [Secure element API](https://github.com/Lora-net/LoRaMac-node/wiki/secure-element) page on the project's wiki for further information.

Please also note that we never had access to a real secure-element device so far.

AlexFBP, 2018-08-30:
Anyways, waht was the last commit that compiled successfully for the the Mote II? Here, we bought 2 of them and all our LoRaWAN devices we have, work mainly on Class A, (we have ~ 20 iM880 and 2 MoteII) so i guess also like @temporaryaccount that it would be better to have a modular code...
If i'm going to work always in a specific class, does it really has sense to compile the other modes?

If you have killed the Mote II :cry: , can you please indicate what was the last commit that compiled successfully for Mote II? Thanks guys...