Lora-net/LoRaMac-node #492: Multiplying repeated MAC commands

XKottelaar, 2018-06-21:
It seems that the MAC commands stored in the MacCommandsBuffer are sometimes being doubled overtime when the node is unable to receive any downlinks. It appears however that the interval at which the commands are doubled is random. 

The MAC commands that are being doubled are the commands that need to be repeated by the node until a downlink has been received. These commands are stored in the MacCommandsBufferToRepeat.

In the PrepareFrame function of the loramac MAC commands  that need to be repeated are added to the MacCommandsBuffer at the MacCommandsBufferIndex. A little further in this function the MacCommandsBuffer is analysed for commands that need to be repeated until a downlink is received. The commands that need to repeated are then placed in MacCommandsBufferToRepeat. The function always does this at index 0 so I see no problem there. However it seems that the problem is an re-entrance problem. The MacCommandsBufferIndex is incremented with the MacCommandsBufferToRepeatIndex. When the MacCommandsBufferIndex has not been reset the commands to be repeated are inserted below the previously repeated commands. The commands in the buffer are now doubled. The doubled commands are then analysed by the function ParseMacCommandsToRepeat()  which now finds the doubled commands and stores them. 

The node continuous to send the doubled commands until they are suddenly doubled again (it doesn’t occur at each uplink). At some point the commands are doubled until we have 8 commands in total. These commands don’t fit in the fOpts field, the node then switches to port 0. At this moment the Application server notices the malfunction as there isn’t any application data send on port 0, only MAC commands. When the node starts sending the MAC commands on port 0 the doubling of the MAC commands stops. We only ever saw a maximum of 8 commands being send on port 0. I think this is because the MacCommandsBufferIndex is set to 0 in the PrepareFrame function when port 0 is used. This makes me suspect that the MacCommandsBufferIndex is sometimes not being reset properly when the node is piggy bagging MAC commands. There are only two places I could find where the MacCommandsBufferIndex is incremented or changed to a non-zero value. One of which is when MacCommands are added to the buffer, this only happens on a downlink (which we don’t receive) or on a link check request (which we don’t use). The other place is in the PrepareFrame(), so I suspect this function to double the MAC commands. 

The buffer is cleared upon receiving a downlink but it is a serious problem when the node is unable to receive the downlinks because it also stops sending application data when the node starts sending on port 0.

Is there any reason of the prepareFrame function to be called multiple times causing this issue or are there other problems here?

We use version 4.4.0. The nodes where we see this issue are sending at most at 5 minute interval (this happens only very rarely and not when we see the doubling of the MAC commands) but they usually send at an interval of an hour. The issues therefore shouldn’t be caused by the dutycycle limitations.

BrandonD200, 2018-06-26:
I have found this issue before. It happens when there are sticky commands to be acked, you are using unconfirmed ULs, and a RX fails (for me the failure was the devEUI received wasn't my devEUI). on RX failure the MacCommandsBufferIndex is not reset and for the next transmission the sticky commands (in MacCommandsBufferToRepeat) are added to the end of MacCommandsBuffer where they are also, causing the duplication. 

My fix was adding:

            if( McpsConfirm.McpsRequest == MCPS_UNCONFIRMED && ChannelsNbRepCounter >= LoRaMacParams.ChannelsNbRep )//if it is unconfirmed and there are no more retransmissions
            {
                MacCommandsBufferIndex = 0; //clear the MacCommandsBuffer
            }

to the OnMacStateCheckTimerEvent() function in the if statement:
 if( ( LoRaMacState & LORAMAC_RX_ABORT ) == LORAMAC_RX_ABORT )

This ensured on rx failure the MacCommandsBuffer was cleared so on the next transmission they would not be duplicated.

XKottelaar, 2018-06-27:
Thank you for your reply!

I can indeed confirm that when the RX of a frame is aborted the MacCommandsBufferIndex isn't reset. This might indeed be the reason why the MAC commands are only sometimes being doubled. 

However I have some more questions, in your fix you verify if the message by the node was unconfirmed but why? Shouldn't the fix be also applied for confirmed frames?

My fix so far is to place the mac commands that need to be repeated in the repeat buffer directly, this is done in the AddMacCommand function. The repeated commands are then added to the mac commands buffer in the prepareFrame function where i check if the same commands are not allready placed in the buffer. I do not use the ParseMacCommandsToRepeat function any more.

This should prevent the commands from being copied and duplicated, however i haven't been able to test this yet.

BrandonD200, 2018-06-27:
When I wrote the fix I determined that there was another path the confirmed packets took that would clear the MacCommandsBufferIndex but I cant seem to find that path now. I will test to see if this issue still happens with confirmed packets.

I think your fix will work, the theory looks sound to me :)

XKottelaar, 2018-06-28:
I have done some testing and it seems that you are right. The duplication only occurs when an unconfirmed uplink is sent and a corrupt downlink is received, which aborts the RX function. 

I found a way to simulate the problem by forcing a repeated MAC response. When the MAC response is added to the buffer and the MacCommandsBufferToRepeatIndex is not 0 I force the abort of a RX frame in the onRadioTxDone function. This makes every downlink received after addint the repeated MAC command to the buffer invalid.

I found that the MacCommandsBufferIndex is being reset for confirmed uplinks in the OnMacStateCheckTimerEvent() (in the part which handles retries).

I've added your fix to the code but also kept mine just in case the duplication of the frames can also be caused by another case.

Now i'm just wondering why the MacCommandsBufferIndex is being reset on so many different places and different conditions, I think this is asking for more trouble down the line when more featues get added to the LoraMAC.

mluis1, 2018-07-03:
We have completely refactored the MAC commands handling. 

Could you please check if it solves the observed issues.

XKottelaar, 2018-07-17:
Looking at the code the issue should be fixed. The sticky MAC commands are only added to the buffer when a request has been received via downlink or requested via application layer. Because the sticky commands arn't read back after adding them to the buffer the commands shouldn't be able to be doubled.

mluis1, 2018-07-17:
Thanks for the feedback. I then close the issue.

closed 2018-07-17 by mluis1

Dollar-Yao, 2019-06-27:
> I have found this issue before. It happens when there are sticky commands to be acked, you are using unconfirmed ULs, and a RX fails (for me the failure was the devEUI received wasn't my devEUI). on [...]

Hi, so you add above solution to where to fix this issue? I use 4.4.1 version

Dollar-Yao, 2019-06-27:
> Looking at the code the issue should be fixed. The sticky MAC commands are only added to the buffer when a request has been received via downlink or requested via application layer. Because the [...]

Hi, you fixed this issue on which version? If I use 4.4.1, so where and how should I modify the code to fix it? Thanks.