Lora-net/LoRaMac-node #495: Send the data once

Ivra23, 2018-06-25:
Hello ,

I use this demo board B-L072Z-LRWAN1 with the soft End Node I-CUBE-LRWAN
But I do not understand why I can not send the data once when I want with function send() or LORA_send( &AppData, LORAWAN_DEFAULT_CONFIRM_MSG_STATE); ? 
and not with a timer (TX_ON_TIMER).

Thank you.

Ivra23, 2018-06-26:
Hello,
No answer to give for a single shipment to TNN server?
thanks.

mluis1, 2018-07-03:
Could you please post your question on ST forums?

This question is related to the I-CUBE-LRWAN which is a port of this project. ST are the right people to ask for support on this.

closed 2018-07-03 by mluis1

Ivra23, 2018-07-03:
Hello,
I posted it on ST but there is nobody who answers me, would not say no answer.
Just remove the sending of several frames, and send once. 

Thank you