Lora-net/LoRaMac-node #497: RTC timer list stoped unexpected

llb126yx, 2018-06-27:
hello:
the code version  i used is V4.3.2.
I just add a timer in mai.c and i start it, here is the code:
```c
static TimerEvent_t CycleTimer;
static void onCycleTimerEvent(void)
{
    TimerStop(&CycleTimer);
    DecreaseMsgTimer(); //
    TimerSetValue( &CycleTimer, 1000);
    TimerStart(&CycleTimer);
}
```
in the first several days, it  work very well.but in some day, it  stopped, but the main process is still runing. I have tested for 3 times, all same.  I do not know why this happen

Ivra23, 2018-06-28:
Hello :)
You know how to send the data once without using the RTC ? thank you for your help.

llb126yx, 2018-07-02:
Thank you sir.  I know how to send the data. But what I whta to say is : the whole lorawan code is based on the RTC timer list, when you need open the RX window, when you need wake up, ie. So i am just curiour why this list will stop?  I am debugging and try to find out.

mluis1, 2018-07-03:
Could you check if the latest versions of the timer.c/h implementations solve your issue.

Since V4.3.2 we have made quite some changes that could solve your issues.
You could also check your application to try to understand why the Timer list becomes empty. If it becomes empty it is most certainly because no new timer has re-started.

The RTC peripheral usage is not mandatory. You may use whatever timer/counter peripheral as time base.
We used the RTC for this purpose because it was the only timer that wasn't stopped when the ST MCU were set in the lowest power modes.

closed 2018-07-03 by mluis1