Lora-net/LoRaMac-node #500: boards drivers changes 4.4.0 to 4.4.1

ChrisEAlfred, 2018-06-30:
We were on v4.4.0 and thought we would update to v4.4.1.

Was it really necessary to change the driver APIs?
Now our existing boards' drivers have to go through major porting changes and full testing.

It should really be possibly to just drop in a new MAC and not touch the drivers.

mluis1, 2018-07-02:
I suppose that you are talking about the radio drivers.
Yes, it was necessary to change the APIs because the SX126x radio addition required some changes to the drivers.

You are right in an ideal world it shouldn't be necessary to change those APIs.

Just to inform you that the next GitHub update (should happen this week) will include a lot more changes to the different APIs. Due to the addition of the secure-element support we had to almost refactor all the MAC layer implementation.  This refactor was also necessary to ease the 1.1.0 firmware update and to reduce to a minimum the changes between 1.0.3 and 1.1.0. We have also completely refactored the rtc-board.c/h, timer.c/h and we have added a new module named systime.c/h.

closed 2018-07-03 by mluis1

ChrisEAlfred, 2018-07-11:
Have the driver API changes been completed?

mluis1, 2018-07-11:
We think that the Radio.h APIs are now completed and shouldn't change.

Although, the SX1272 and SX1276 drivers implementation may change in the future. We would like to make their construction similar to the SX126x drivers.

The other APIs (ie: MAC layer) are now quite stable as well. They shouldn't change in a near future unless we have to fix a potential unknown issue.

Please note that some changes happened since 4.4.1 release.
The preparation work for 4.4.2 release is currently available on develop branch. (LoRaWAN 1.0.3)
The preparation work for 5.0.0 release is currently available on feature/5.0.0 branch. (LoRaWAN 1.1.0/1.0.3)

We are currently fixing the last known issues on develop branch (some of the issues are described in GitHub and some others on our internal git server). Once this work is finished we will start sending the binary images to the different test houses for validation.