Lora-net/LoRaMac-node #504: Issue in Cmake and Build process

manii120, 2018-07-05:
Hi friends,

Currently i am facing an issue related to LoRaWAN code make and build process. \
What am i doing right now is,
I downloaded Cmake, GCC-arm , and minGW exe files and gave their path in system variables setting accordingly.
But when i try to make and build LoRa code it shows an error something like this-
[image]
I tried several times with different solutions provided on internet but still unable to solve this issue.
Kindly help me please if any further info needed please let me know.

Thanks

mluis1, 2018-07-05:
The setup of these tools on Windows are quite tricky. It is a lot easier for Linux or MacOS

But, if you carefully follow the provided instructions on (Development Environement)[https://github.com/Lora-net/LoRaMac-node/blob/develop/Doc/development-environment.md] document you should be able to have a working environment.

That said. Could you please provide the full `cmake` command that you are issuing?

My understanding is that the `-DTOOLCHAIN_PREFIX` and `-DCMAKE_TOOLCHAIN_FILE` parameters are most certainly missing.

For instance to build the ClassA example for NucleoL073 platform using SX1262DVK1CAS mbed shield the following commands should be issued.

```bash
MSYS /c/loramac-node$
MSYS /c/loramac-node$ mkdir build
MSYS /c/loramac-node$ cd build
MSYS /c/loramac-node/build$ cmake -DCMAKE_BUILD_TYPE=Debug \
           -DTOOLCHAIN_PREFIX="/c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major" \
           -DCMAKE_TOOLCHAIN_FILE="../cmake/toolchain-arm-none-eabi.cmake" \
           -DAPPLICATION="LoRaMac" \
           -DCLASS="classA" \
           -DCLASSB_ENABLED="OFF" \
           -DACTIVE_REGION="LORAMAC_REGION_EU868" \
           -DREGION_EU868="ON" \
           -DREGION_US915="OFF" \
           -DREGION_CN779="OFF" \
           -DREGION_EU433="OFF" \
           -DREGION_AU915="OFF" \
           -DREGION_AS923="OFF" \
           -DREGION_CN470="OFF" \
           -DREGION_KR920="OFF" \
           -DREGION_IN865="OFF" \
           -DREGION_RU864="OFF" \
           -DBOARD="NucleoL073" \
           -DMBED_RADIO_SHIELD="SX1262DVK1CAS"  ..

-- The C compiler identification is GNU 7.2.1
-- The CXX compiler identification is GNU 7.2.1
-- Check for working C compiler: /c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working C compiler: /c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Linker script: /c/loramac-node/src/boards/NucleoL073/cmsis/arm-gcc/stm32l073xx_flash.ld
-- The ASM compiler identification is GNU
-- Found assembler: /c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /c/loramac-node/build

MSYS /c/loramac-node/build$ make -j 4

Scanning dependencies of target peripherals
Scanning dependencies of target NucleoL073
Scanning dependencies of target radio
Scanning dependencies of target system
[  1%] Building C object src/radio/CMakeFiles/radio.dir/sx126x/radio.c.obj
[  2%] Building C object src/peripherals/CMakeFiles/peripherals.dir/gpio-ioe.c.obj
[  3%] Building C object src/system/CMakeFiles/system.dir/adc.c.obj
[  5%] Building C object src/radio/CMakeFiles/radio.dir/sx126x/sx126x.c.obj
[  6%] Building C object src/system/CMakeFiles/system.dir/delay.c.obj
[  7%] Building C object src/peripherals/CMakeFiles/peripherals.dir/mag3110.c.obj
[  8%] Building C object src/system/CMakeFiles/system.dir/eeprom.c.obj
[  8%] Built target radio
[ 10%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/board.c.obj
[... 60 lines cut ...]
[ 83%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_i2c_ex.c.obj
[ 83%] Built target mac
[ 84%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_pwr.c.obj
[ 86%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_pwr_ex.c.obj
[ 87%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rcc.c.obj
[ 88%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rcc_ex.c.obj
[ 89%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rtc.c.obj
[ 91%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rtc_ex.c.obj
[ 92%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_spi.c.obj
[ 93%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_uart.c.obj
[ 94%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_uart_ex.c.obj
[ 96%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/sx1262dvk1cas-board.c.obj
[ 96%] Built target NucleoL073
Scanning dependencies of target LoRaMac-classA
[ 97%] Building C object src/apps/LoRaMac/CMakeFiles/LoRaMac-classA.dir/common/NvmCtxMgmt.c.obj
[ 98%] Building C object src/apps/LoRaMac/CMakeFiles/LoRaMac-classA.dir/classA/NucleoL073/main.c.obj
[100%] Linking C executable LoRaMac-classA
   text    data     bss     dec     hex filename
  74536     408    6928   81872   13fd0 LoRaMac-classA
[100%] Built target LoRaMac-classA
Scanning dependencies of target LoRaMac-classA.hex
Scanning dependencies of target LoRaMac-classA.bin
[100%] Built target LoRaMac-classA.hex
[100%] Built target LoRaMac-classA.bin
```

manii120, 2018-07-07:
Thanks for your reply, and it helped me to configure it properly but when i use make -j 4 command it shows - "no command found or recognized". please help i'm able to get make file in build folder but unable to follow next step.

And above my only mistake is using '\' this slash instead of '/'. but now its working. Thanks to you.

manii120, 2018-07-09:
Yes thank you so much. Finally found an solution by installing gnuwin which is make utility for windows and after giving its path finally now i am able to make my project. Thanks again.

closed 2018-07-09 by mluis1