Lora-net/LoRaMac-node #505: RF_IDLE when trying to open RX2 window

ananevilya, 2018-07-05:
RegionXXXXXRxConfig called from OnRxWindow2TimerEvent always return RF_IDLE .  And RX2 window can't be opened.
I'm testing 5.0.0 version

mluis1, 2018-07-05:
Is the hardware platform that you are using one natively supported by this project?
If it is not the case could you please verify that you haven't missed something when upgrading.

Anyway, we will investigate this as soon as possible.

Although, please note that we have run the LoRaWAN 1.0.2 specification ClassA EU868 and US195 certification tests with branch 5.0.0 code and almost all tests succeeded. Those tests verify that the end-device correctly handles Rx1 and Rx2 windows in plus a PER is performed on the 2 windows for each possible datarate.

The failing tests are mainly due to the LoRaWAN 1.1 new way of doing the uplink/downlink re-transmissions.

The only known real issue that we have observed concerns the JoinReq procedure. While performing the OTAA tests the test tool doesn't answer to the JoinReq on purpose and sometimes the MAC layer reports it is "busy" and never gets out of this state. This issue happens randomly which makes the debug process hard to perform and time consuming.

ananevilya, 2018-07-05:
Sorry, this is really my porting issue.
I'm not change one of port name of used pin connected to sx1272.
Thanks.

closed 2018-07-05 by ananevilya