Lora-net/LoRaMac-node #506: SX1276 radio driver power handling issue

mluis1, 2018-07-06:
The RFO `OutputPower` bits aren't correctly handled.

The `Pmax` value is set to 15 dBm in `void SX1276SetRfTxPower( int8_t power )` function at line:
```c
paConfig = ( paConfig & RF_PACONFIG_MAX_POWER_MASK ) | 0x70;
```
This setting makes that `OutputPower` bits must range from 0 up to 15.

The `if( power < -1 )` check was most certainly introduced because the `Pmax` radio hardware default value was 13.2 which makes `OutputPower` range from -1.8 up to 13.2

The solution is to change the `if( power < -1 )`  check into:
```c
        if( power < 0 )
        {
            power = 0;
        }
        if( power > 14 )
        {
            power = 14;
        }
```

GrumpyOldPizza, 2018-07-06:
My hunch is that the reason for this is that the code simply got copied from SX1272.

Our modified code uses this here:

        if( power > 0 )
        {
            if( power > 15 )
            {
                power = 15;
            }

            paConfig = ( paConfig & RF_PACONFIG_MAX_POWER_MASK & RF_PACONFIG_OUTPUTPOWER_MASK ) | ( 7 << 4 ) | ( power - 0 );
        }
        else
        {
            if( power < -4 )
            {
                power = -4;
            }

            paConfig = ( paConfig & RF_PACONFIG_MAX_POWER_MASK & RF_PACONFIG_OUTPUTPOWER_MASK ) | ( 0 << 4 ) | ( power + 4 );
        }

But then again, I have no idea how to measure or model the path loss. Suppose this has to be done for both RFO and PABOST per board ...

Don't know whether it's worth it. For US915, I'd actually want 10dbm to be lower to save power. On the other hand, for EU868, I'd want 14dbm to be really 14dbm (or 15dbm) to compensate for the path loss.

mluis1, 2018-07-06:
These functions are platform specific and one can customize them in order to have the best performances in accordance to the target hardware.

On the provided examples we will make the proposed change in order to be coherent.

mluis1, 2018-07-06:
I have applied your recommendation.
Could you please review the changes a let me know if you are OK with it.

```c
void SX1276SetRfTxPower( int8_t power )
{
    uint8_t paConfig = 0;
    uint8_t paDac = 0;

    paConfig = SX1276Read( REG_PACONFIG );
    paDac = SX1276Read( REG_PADAC );

    paConfig = ( paConfig & RF_PACONFIG_PASELECT_MASK ) | SX1276GetPaSelect( SX1276.Settings.Channel );

    if( ( paConfig & RF_PACONFIG_PASELECT_PABOOST ) == RF_PACONFIG_PASELECT_PABOOST )
    {
        if( power > 17 )
        {
            paDac = ( paDac & RF_PADAC_20DBM_MASK ) | RF_PADAC_20DBM_ON;
        }
        else
        {
            paDac = ( paDac & RF_PADAC_20DBM_MASK ) | RF_PADAC_20DBM_OFF;
        }
        if( ( paDac & RF_PADAC_20DBM_ON ) == RF_PADAC_20DBM_ON )
        {
            if( power < 5 )
            {
                power = 5;
            }
            if( power > 20 )
            {
                power = 20;
            }
            paConfig = ( paConfig & RF_PACONFIG_OUTPUTPOWER_MASK ) | ( uint8_t )( ( uint16_t )( power - 5 ) & 0x0F );
        }
        else
        {
            if( power < 2 )
            {
                power = 2;
            }
            if( power > 17 )
            {
                power = 17;
            }
            paConfig = ( paConfig & RF_PACONFIG_OUTPUTPOWER_MASK ) | ( uint8_t )( ( uint16_t )( power - 2 ) & 0x0F );
        }
    }
    else
    {
        if( power > 0 )
        {
            if( power > 15 )
            {
                power = 15;
            }
            paConfig = ( paConfig & RF_PACONFIG_MAX_POWER_MASK & RF_PACONFIG_OUTPUTPOWER_MASK ) | ( 7 << 4 ) | ( power );
        }
        else
        {
            if( power < -4 )
            {
                power = -4;
            }
            paConfig = ( paConfig & RF_PACONFIG_MAX_POWER_MASK & RF_PACONFIG_OUTPUTPOWER_MASK ) | ( 0 << 4 ) | ( power + 4 );
        }
    }
    SX1276Write( REG_PACONFIG, paConfig );
    SX1276Write( REG_PADAC, paDac );
}
```

GrumpyOldPizza, 2018-07-06:
Looks good.

- Thomas

mluis1, 2018-07-06:
Thanks. The repository will be updated at the next push.

closed 2018-07-06 by mluis1