Lora-net/LoRaMac-node #508: VS code Debug err !!

zhoubx123, 2018-07-09:
hello,

when i use command Debug, there

[cms-client] None of preferred generators available on the system.
[rollbar] Unhandled exception: Unhandled Promise rejection: Changing build variant Error: Unable to determine CMake Generator to use {}
[cms-client] None of preferred generators available on the system.
[rollbar] Unhandled exception: Unhandled Promise rejection: setVariant Error: Unable to determine CMake Generator to use {}

who can help me .

mluis1, 2018-07-09:
Please follow the instructions provided on [Development Environement](https://github.com/Lora-net/LoRaMac-node/blob/develop/Doc/development-environment.md) document.

Please note that before hitting the Debug command you must first successfully build your firmware.

As the information that you provide isn't very clear I suppose that you are on a Windows environment.

### Configuration

For Windows platforms it is necessary to make some additional configurations. Open your settings under *File->Preferences->Settings* and add the following lines:

Add MinGW Makefiles as preferred Generator:

```json
"cmake.preferredGenerators": [
        "MinGW Makefiles",
        "Ninja",
        "Unix Makefiles"
    ]
```

zhoubx123, 2018-07-09:
o my god , some else make mistake, now can woke correct.

thank you!

closed 2018-07-09 by zhoubx123