Lora-net/LoRaMac-node #510: In the develop branch, posible race condition on IsMacProcessRequired in main.c

mdoreille, 2018-07-10:
In the development branch of LoRaMac-node/src/apps/LoRaMac/classA/NucleoL073/main.c on line 1155, 
the IsMacProcessRequired == 0 test must be performed within the __disable_irq() section of the BoardLowPowerHandler function. Otherwise, a call of the OnMacProcessNotify function can occur from an ISR between the positive test on IsMacProcessRequired == 0 and the entrance of the MCU in sleep or stop mode.

mluis1, 2018-07-12:
Yes, you are right. We will propose a solution as soon as possible.

In the mean time one can change the code from:
```c
                if( IsMacProcessRequired == 0 )
                {
                    // Wake up through events
                    BoardLowPowerHandler( );
                }
```

to something like:
```c
                CRITICAL_SECTION_BEGIN( );
                if( IsMacProcessRequired == 0 )
                {
                    // Wake up through events
                    BoardLowPowerHandler( );
                }
                CRITICAL_SECTION_END( );
```

mluis1, 2018-07-18:
Could you please check the proposed fix?

If you are OK with it could you please close the issue?

Thanks in advance.

mdoreille, 2018-07-19:
By looking at the code, the problem should be fixed.

closed 2018-07-19 by mdoreille