Lora-net/LoRaMac-node #512: Timer at 1 second interval

ChrisEAlfred, 2018-07-16:
What would be the correct way to get a timer event once per second (on average will be 1Hz).
We are using the STM32L083 (similar to STM32L073)

Is it correct to do something like:
`
static void OnTickTimerEvent(void)
{
    TimerStop(&TickTimer);
    
    // Capture current time
    TimerTime_t currentTime = TimerGetCurrentTime();
    
    // Schedule a timer for the next second boundary
    TimerTime_t nextTickTime = (currentTime - (currentTime % 1000)) + 1000;
    
    // Calculate time to next interval
    TimerTime_t nextTickTime - currentTime;

    TimerSetValue(&TickTimer, nextTickInterval);
    TimerStart(&TickTimer);

    lastTickTime = nextTickTime;
}
`

Or will this have a problem at the boundary of the RTC rolling over after the 49 days in rtc-board.c (see git commit 1968e2d080585ebc8dddb3eb28f7e427e253caee for Motell).

mluis1, 2018-07-16:
The proposed solution looks to be quite complex.

In order to generate a continuous 1 second tick I would do it as follows:

```c
static void OnTickTimerEvent(void)
{
    TimerStop(&TickTimer);
    
     // Set timer for 1 more second and restart it.
    TimerSetValue( &TickTimer, 1000 );
    TimerStart( &TickTimer );
}
```

The hardware timer overflow is handled in the lower layers.

ChrisEAlfred, 2018-07-16:
Does this have a small cumulative error between the TimerStop and TimerStart such that the interval is actually 1000ms + small amount.

Is it ok to to have some dither, but I can not have a long term drift.

mluis1, 2018-07-16:
Yes, the interruption handling plus TimerStop/TimerStart takes some time. If you require to have very precise time, then, you need to characterize how much overhead the interrupt handling plus TimerStop/TimerStart introduces. Once, the characterization is done you should get the amount of milliseconds that are required to be compensated.

Depending on what you are trying to achieve there are maybe other solutions.

ChrisEAlfred, 2018-07-17:
I can't see how characterisation can be done as the interrupts are random external events.

The software timer should add a mean zero error to the error of the crystal used to drive the RTC.
Maybe using the Timer routines (milliseconds) is not the right solution because of the 0x7D000000 rollover (see Motell rtc-board.c 49 day fix).

Is there a long term accurate way of accumulating time e.g. using the RTC 'tick'?

mluis1, 2018-07-17:
What I meant by characterization was that one could compute the time required to execute the interruption plus the time between TimerStop/TimerStart. This time is more or less constant (depends on how much conditions are used in the code). So, once the characterization is done you need to adjust the 1000 milliseconds value by subtracting the characterized time value.

That said, I guess that I don't really understand your issue.
If the issue is just to keep the current date/time value accurate then in develop branch we have refactored the RTC handling and we have added a sysTime.c/h module which handles the UNIX/GPS time.

The `SysTimeGet` function will return the following structure:
```c
/*!
 * \brief Structure holding the system time in seconds and milliseconds.
 */
typedef struct SysTime_s
{
    uint32_t Seconds;
    int16_t  SubSeconds;
}SysTime_t;
```

The `SysTimeGet` function returns the current number of seconds/sub-seconds since UNIX epoch origin.
The `SysTimeSet` function allows the update of the current number of  seconds/sub-seconds since UNIX epoch origin.

The `SysTime_t` structure overflows after around 136 years which should be enough for the product life time.

The LoRaWAN 1.0.3 specification has a new MAC command to ask the network server for the current time. Like this the end-device can synchronize its time with network server time.

ChrisEAlfred, 2018-07-17:
Thanks for your quick response.

We feel it is best to move to the latest develop branch. Bit of a big change to the custom board drivers, but worth it.

closed 2018-07-17 by ChrisEAlfred