Lora-net/LoRaMac-node #513: Stack porting on Kinetis KM334

BenNat974, 2018-07-16:
Hey, 

I implemented Loramac Stack on Kinetis MKM34Z256VLQ7 provided by NXP. I use Sx1272MB2DAS provided by Semtech to communicate.  I want to give you feedbacks about the porting. 

Porting of spi-board, gpio-board, sx1272-board was has been done without great difficulty. Porting of rtc-board has been more challenging.

Several µc supported by the stack use RTC wich are able to manage milliseconde precision. My RTC has only a precision of seconde. For this reason, I use a LPTMR on cascade with RTC. Both are synchronous and permit to save a seconde and milliseconde context every time. Moreover, LPTMR operate also on several sleep mode. With this architecture, I valided class A and Class C communication. 

This adaptation is binding  due to the usage of two peripherals which must be manage with a perfect synchronization. Moreover, that induces critical zone. between interupt routine and TimerIrqHandler.

Have you ever been confronted with the portage of the stack on µc without RTC with milliseconde precision ? Have you a different solution to this problematic ? In the future, will there be an evolution of the stack for this µc profil ?

Thanks for you work,
Benjamin

jamesl-dm, 2018-07-16:
I also think a better approach might be to have a coarse and a fine sleep. Use the coarse sleeps for duty cycle backoffs, and the fine sleeps for the tightly controlled radio rx windows. Then the board can use a regular (high power) timer for the fine sleeps if it needs to. But I guess there is nothing stopping implementors from doing this with the current timer API - they could just assume that any sleep shorter than 10 seconds was a fine sleep.

mluis1, 2018-07-16:
Thanks for the feedback and well done doing the porting to the Kinetis.

The usage of an RTC peripheral is not mandatory. You can use whatever hardware timer your MCU provides.

We used the RTC peripheral because on STM32 MCUs it was the only timer that was able to operate and wake-up the MCU from the lowest power modes. Nowadays there are STM32 MCUs that also support a Low power timer.

The only requirement for the stack is to have a 32 bits free running timer with the possibility to set a comparison value which when equal to the timer value generates an interrupt able to wake-up the MCU.
In order to improve the timer precision the timer input clock should be a 32kHz crystal oscillator. This is most important in case you want to use ClassB.

Thus, in your case you could just use the LPTMR peripheral and counting milliseconds.

On develop branch we have updated the STM32 RTC handling in order to simulate a 32 bits free running timer. The sysTimer.c/h files are there to handle the real time usage.

Hope this clarifications help you.

BenNat974, 2018-07-16:
Unfortunately, Kinetis Km34 don't have LPTimer/Timer with 32 bits. They have only 16bits. With this hardware limitation, I think my adaption is the best solution.

mluis1, 2018-07-16:
I think that you may also use the 16 bits timer.

Personally I don't know the Kinetis but, I suppose that the HW timer can generate an overflow interrupt.
If so, you could do something similar to the code shown below. (based on SAML21 rtc-board.c file)

As you can see, every time the 16 bits HW timer overflows the `RtcOverflowIrq` callback should be called and we just increment the 32 bits global variable handling the time by, 1 left shifted by 16 bits.
```c
static void RtcOverflowIrq( void )
{
    RtcTimerContext.Time += ( uint32_t )( 1 << 16 );
}
```

Please note that the `RtcProcess` function may not be necessary in your case.

Modified SAML21 rtc-board.c

```c
// Add required include files
#include "rtc-board.h"

#define MIN_ALARM_DELAY                             3 // in ticks

/*!
 * \brief Indicates if the RTC is already Initialized or not
 */
static bool RtcInitialized = false;
static volatile bool RtcTimeoutPendingInterrupt = false;
static volatile bool RtcTimeoutPendingPolling = false;

typedef enum AlarmStates_e
{
    ALARM_STOPPED = 0,
    ALARM_RUNNING = !ALARM_STOPPED
} AlarmStates_t;

/*!
 * RTC timer context 
 */
typedef struct
{
    uint32_t Time;  // Reference time
    uint32_t Delay; // Reference Timeout duration
    uint32_t AlarmState;
}RtcTimerContext_t;

/*!
 * Keep the value of the RTC timer when the RTC alarm is set
 * Set with the \ref RtcSetTimerContext function
 * Value is kept as a Reference to calculate alarm
 */
static RtcTimerContext_t RtcTimerContext;

/*!
 * Used to store the Seconds and SubSeconds.
 * 
 * WARNING: Temporary fix fix. Should use MCU NVM internal
 *          registers
 */
uint32_t RtcBkupRegisters[] = { 0, 0 };

/*!
 * \brief Callback for the hw_timer when alarm expired
 */
static void RtcAlarmIrq( void );

/*!
 * \brief Callback for the hw_timer when counter overflows
 */
static void RtcOverflowIrq( void );
[... 157 lines cut ...]
        }
    }
    CRITICAL_SECTION_END( );
}

TimerTime_t RtcTempCompensation( TimerTime_t period, float temperature )
{
    return period;
}

static void RtcAlarmIrq( void )
{
    RtcTimerContext.AlarmState = ALARM_STOPPED;
    // Because of one shot the task will be removed after the callback
    RtcTimeoutPendingInterrupt = false;
    // NOTE: The handler should take less then 1 ms otherwise the clock shifts
    TimerIrqHandler( );
}

static void RtcOverflowIrq( void )
{
    RtcTimerContext.Time += ( uint32_t )( 1 << 16 );
}
 
```

BenNat974, 2018-07-17:
Your example use an non volatile ram to save time context. My board is not equipped with this type of memory space.

mluis1, 2018-07-17:
The provide example uses a normal global variable. The SAML21 platform has no NVM memory support yet.

```c
/*!
 * Used to store the Seconds and SubSeconds.
 * 
 * WARNING: Temporary fix fix. Should use MCU NVM internal
 *          registers
 */
uint32_t RtcBkupRegisters[] = { 0, 0 };
```

BenNat974, 2018-07-17:
With you implementation, low power mode  is it supported ?

mluis1, 2018-07-17:
Yes, it should. The only constraint for you as you don't have a way to store the the time on a NVM memory is that the MCU must keep the RAM memory active in the lower power mode.

As I don't know very well your MCU I just took a quick read at its [reference manual](https://www.nxp.com/docs/en/reference-manual/KM34P144M75SF0RM.pdf) and in chapter 38.2 it is said the following:
>This block is a low power module that provides time keeping and calendaring functions
and additionally provides protection against tampering (external or internal tamper
events), protection against spurious memory/register updates and battery operation. A
Standby RAM is provided if the CPU wants to store any data that has to be retained when
in battery operation mode.

This means that this MCU has also ways to store data that must be retained. You have at your disposal 32 bytes of backed up RAM.

Chapter 38.2.1 - page 718
>32 Bytes of Battery Backed Up RAM

Another possibility for you would be to use an external I2C EEPROM to store the different contexts. Please note that in future storing MAC contexts will be mandatory (LoRaWAN specification 1.1).

It should also be possible to use part of the MCU FLASH memory as a NVM memory. Please check the [reference manual](https://www.nxp.com/docs/en/reference-manual/KM34P144M75SF0RM.pdf)l and MCU provided application notes. I'm quite confident that something exists explaining how to do it.

closed 2018-07-26 by mluis1