Lora-net/LoRaMac-node #514: Power off radio inbetween class A transmit/receive sesions

valerianrusu, 2018-07-23:
I use the stack in a class A application that is sending data very seldom. In order to reduce power consumption I would like to power off the radio chip in-between transmit/receive session. 
Looking at version 4.4.1 of the stack I was not able to identify a simple way to have this implemented. It would be nice if the stack would provide support so that the radio will be powered up and reinitialized when a frame is requested to be sent then powered down after the receive window. Or at least to provide an API to reinitialize the radio and to be able to register a callback to be informed when receive window is over.

mluis1, 2018-07-24:
I'm not sure to understand the issue but, the provided examples already handle the low-power management.

The measured power consumption on STM32 based platforms is around 1.2 µA when the low-power mode is activated.

All the APIs and Callbacks required to handle the low power modes are in place.

tititin42, 2018-08-01:
Same issue here. There is no 'obvious' ways in the provided stacks where we can turn off the chip and make it back on again.
This possibility is not offered by your devices, but I'm using the stack in a different environment, and I'm able to shut down the chip and on again (instead of placing it to sleep mode)

closed 2019-02-12 by mluis1