Lora-net/LoRaMac-node #516: Why default data rate range is between DR_0 and DR_5 in applying CFList? Why not EU868_TX_MIN_DATARATE and EU868_TX_MAX_DATARATE?

MaheshKumarPN, 2018-07-24:
I see form the RegionXXXX.c files that RegionXXXXApplyCFList() api definition has default data rate to be DR_0 and DR_5 hard coded.
Do we have a way now to update them form application context?
Do we have any future plan to move it to application context?

or Am i missing something?

mluis1, 2018-07-24:
The 3 default channels are defined with a range between DR_0 to DR_5.
For EU868 the regional parameters (LoRaWAN Regional Parameters v1.0.3rA) table 3 shows this:

[image]

Please note that it is up to the network server to then instruct the end-device to enable the upper datarates.

The NewChannelReq MAC command provides the way to change the datarate range. (LoRaWAN Specification v1.0.3)

[image]

mluis1, 2018-07-24:
I forgot to mention that the DR_6 and DR_7 datarates use a specific modems on the SX1301. These modems have independent channel frequencies parameters.

This means that there is no way for an end-device to figure out which frequencies are being used. This is one of the reasons why the Network Server must create/change the channels on the end-device.

closed 2019-02-12 by mluis1