Lora-net/LoRaMac-node #517: Suggestion: system drivers callbacks should have user defined data passed back

ChrisEAlfred, 2018-07-24:
It would be useful to add a user parameter to be passed as the argument in callbacks.
For example in the timer:

`// In timer.h
typedef struct TimerEvent_s
{
    ...
    void ( *Callback )( void *context ); //! Timer IRQ callback function WITH context
    ...
    void *Context;  //! Context to pass back

}TimerEvent_t;

// In timer.c
#define ExecuteCallBack( _callback_, context ) \
    do                                \
    {                                 \
        if( _callback_ == NULL )      \
        {                             \
            while( 1 );               \
        }                             \
        else                          \
        {                             \
            _callback_( context );    \
        }                             \
    }while( 0 );

void TimerInit( TimerEvent_t *obj, void ( *callback )( void *context ), void *context )
{
    ...
    obj->Context = context;
}
`

Also applies to gpio GpioIrqHandler and GpioIoeIrqHandler.
And to a lesser degree uart, as it already passes a uart 'identifier'

ChrisEAlfred, 2018-07-24:
This is useful for passing context (e.g. 'this' in C++).

mluis1, 2018-07-26:
Thanks for the proposition, we will discuss it internally and see how we can add it.

closed 2018-10-26 by djaeckle